MarlinFirmware/Marlin #11735: TMC2130 and dual motors on the z axis with wrong directions

oangelo, 2018-09-06:
I'm having a very strange problem. Using the TMC2130 (SPI) and two motors on the z axis, a very strange behavior happens. When the z-axis should rise, each motor rotates in opposite directions, bending the x-axis. When the z-axis must be lowered, both motors rotate in the same direction and the x-axis descends correctly. I already tried both versions of bugfix (1.1, 2.0) without lucky.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

ManuelMcLure, 2018-09-06:
Are both motors connected to the same driver or are they on separate drivers?

RudyBenner, 2018-09-06:
Could you reverse the motor connections on one motor?

oangelo, 2018-09-06:
@ManuelMcLure , I have separate drivers. @RudyBenner , reverse the motor connections will fix one direction (up) and mess the other (down). I was using DRV8825 before, and it was working fine. I guess the problem is related with TMC2130.

teemuatlut, 2018-09-06:
> reverse the motor connections will fix one direction (up) and mess the other (down). 

Flip only one.

kAdonis, 2018-09-06:
It looks like, one of your drivers is not receiving the direction signal
Check the pinheaders at the TMC2130. Are they soldered properly? Especially the DIR pin

oangelo, 2018-09-09:
Indeed, one of the drivers dir pin was not working. Thanks for the support guys.

closed 2018-09-09 by oangelo