MarlinFirmware/Marlin #11739: Linear Advance step generation issue causing motor noise. Logic Analyser proof included.

SJ-Innovation, 2018-09-06:
### Description

Hi everyone, following on from some other issues with linear advance i did some probing with a logic analyser on the Step and Dir pins to the extruder motor. I found some very strange behaviour that i cant fully explain which is causing a lot of noise on the motor.

As seen in the attached logic analyser dumps, with linear advance disabled the step pulse generation is very even and accelerated correctly with no missed or duplicated steps. With linear advance enabled (settings below) there are frequency missed and duplicated step pulses. While this doesnt affect the print quality, it does significantly increase the amount of noise and mechanical wear on the extruder motor gears as every missed or duplicated pulse creates a momentary high-jerk motion. 

LogicAnalyser Dumps can be opened and navigated with PulseView (free)
Photos of the effects are included, but viewing in PulseView will give allow you to measure timings etc. 

### Steps to Reproduce

Linear Advance Enabled - Must
S-Curve Enabled / Disabled - Makes no difference
Junction Dev - Makes no difference

Hardware:
Mega 2560 Running Marlin 2 Dev Branch
RAMPS
TMC2130 on all axis - Though i cant see it would make a difference, this being pulse gen related. 

Linear Advance K > 0, Mine is about k=0.7 due to Bowden Tubes. 

Im investigating what the cause could be, im convinced its somewhere in the Advance ISR, but may be totally wrong. Would love a little bit of assistance. 

Thanks guys! 
Sam

[LinAdv LogicDump.zip](https://github.com/MarlinFirmware/Marlin/files/2356711/LinAdv.LogicDump.zip)
[image]
[image]

Itox001, 2018-09-06:
Try the fix @Sebastianv650 just provided on my LA issue #11711. It's a fix to the more extreme inconsistent extrusion behavior seen and it's actually related to the advance ISR as you suggested. Probably yours is a less severe manifestation of the same issue.

I haven't had the chance to test it, but will do so later today.

SJ-Innovation, 2018-09-06:
Thanks,
I’ll try the fix tomorrow, though running the logic through based on my understanding of the advance ISR I think this should fix it. Though the acid test will be tomorrow when I test it. More to come.

Thanks :)

SJ-Innovation, 2018-09-07:
Test Results:
No Change, extruder motor still generates just as much noise with the fix as without, and logic analyser dump looks exactly the same with frequent missed and double pulses. 

Thanks for the suggestion though, it was worth a shot.

AnHardt, 2018-09-07:
Sadly your dumps are telling just a tiny little bit of the story. Without the context the dumps have no value.
What channels do we see?
If E_STEP and E_DIR, what do the other axis look like? What is the "leading" axis?
What g-code do we see?
What is your active configuration?

"Linear Advance" does add and omit steps to add or reduce pressure - that's it's job. Seeing them (at the right places) means - it's working.

SJ-Innovation, 2018-09-07:
I know theyre telling a tiny bit of the information, as for having no value... They have more value than if they weren't included. 

As described above OP, the dumps have the "Step and Dir pins to the extruder motor". The other axis are irrelevant, as the EMotor is handled in its own ISR and doesnt care about the leading axis, only the number of steps the block has, and has left. 

I know LinAdv adds or omits steps ive been using it for a long time. Take a closer look at the dumps and youll notice the issue im describing. Missing or duplicating pulses in an otherwise constant speed, or constant acceleration move results in an audible harmonic at that frequency instead of a smooth acceleration or deceleration.

Did you view the dumps in Pulseview as i suggested? Or are you simply looking at the pictures. If only looking at the pictures then of course there isnt much information. In Pulseview using a "Stepper Motor" classifier with the D0 and D1 channels as step and dir will output the current step rate and position. As seen, with LinAdv disabled: It is a smooth progression as the move takes place (This is a part of an accelaration) 
[image]

Whereas with LinAdv Enabled: Notice the regular gaps.
[image]

Once again dont rely on these images, use Pulseview to see for yourself.

SJ-Innovation, 2018-09-07:
Okay! More testing done! This time including data from all of the axis over a much longer period of time. I recommend not decoding the Z axis moves as Pulseview crashes due to running out of memory. 

[LinAdv k=0.7.zip](https://github.com/MarlinFirmware/Marlin/files/2361131/LinAdv.k.0.7.zip)

Live deductions made:
What strikes me as strange, is that every time an EStep pulse is seemingly "missed", the EDir line is inverted until the next pulse. Tracing this in the code, I see this results from LA_steps briefly becoming >=0 from being <0. This means that the block containing line 1803 (inclusive, stepper.cpp) fired before it (to increment LA_steps from -1 to 0); this means that (based on that its prior if statement didnt fire) we havnt had enough steps to start decelarating and that we havnt yet advanced all the way to where we need to be.
I think there may be some race condition involved. Perhaps the one described in stepper.cpp lines 1227 onwards? 

As for the leading axis, its always either X or Y.

SJ-Innovation, 2018-09-08:
Fixed, issue with LinAdv cruise phase, fixed by pull 11759. Coupled with low extruder steps per mm exacerbating harmonic issue.

closed 2018-09-08 by SJ-Innovation