MarlinFirmware/Marlin #11740: Compiling 2.x with platform io for anet

aplocher, 2018-09-06:
The 2.x branch doesn't seem to have anet10 defined in the platformio.ini, but the 1.x branches do.  I'm not sure if there's a new recommended way to use platformio for anet in the 2.x branch or if this was just overlooked, but I've modified the 1.x branch's definition just slightly and when I add it to platformio.ini it seems to work well.

Just wondering, is there a different way I should be compiling for anet devices in the 2.x branch?

Or is this worthy of a pull request?

```
#
# ANET
#
[env:anet10]
platform = atmelavr
framework = arduino
board = sanguino_atmega1284p
build_flags = ${common.build_flags}
upload_speed = 115200
lib_deps = ${common.lib_deps}
```

Then using `platformio run -e anet10` and `platformio run -e anet10 -t upload` will work.

Thanks guys!

thinkyhead, 2018-09-06:
That environment was removed and isn't in `1.1.x` or `bugfix-1.1.x` anymore either. You should now be using `sanguino_atmega1284p` or the Auto Build option.

closed 2018-09-06 by thinkyhead

aplocher, 2018-09-06:
Thank you, I kinda assumed I was just doing something wrong.  I copied it from the other branch several months ago (when it was still there, I guess).  I just tried sanguino_atmega1284p and it worked just fine.  Thanks again!