MarlinFirmware/Marlin #11741: Update 1.1.8 to 1.1.9 Auto Bed Leveling Bilinear Problem

rsodvd79, 2018-09-06:
Hi, I have a problem after updating Merlin from version 1.1.8 to version 1.1.9.
I have used the same parameters from the Configuration.h and Configuration_adv.h files of the older version,
but the AUTO_BED_LEVELING_BILINEAR behaves badly with the new firmware.
Homing and axis movements work correctly with the new firmware, the problem occurs when I try the bed leveling, the y axis moves to position 0, but when it has to increase it does not work, it seems that the stepper motor runs in the opposite direction.
My printer is a HICTOP 3DP11 Prusa i3.

In this video we see the behavior of the printer.
https://www.youtube.com/watch?v=O6S0Xgt43j0

cjsoong, 2018-09-06:
@rsodvd79  
Configuration.h
#define DEFAULT_AXIS_STEPS_PER_UNIT {80, 400, 400, 94.5} ...Error
The number of steps in Y is wrong. It should be 80 like X.
#define DEFAULT_AXIS_STEPS_PER_UNIT {80, 80, 400, 94.5}

rsodvd79, 2018-09-06:
400 is right because I have the screw on the y axis

rsodvd79, 2018-09-07:
I have set
#define DEFAULT_MAX_FEEDRATE {300, 100, 5, 25}
and
#define DEFAULT_MAX_ACCELERATION {1000, 100, 100, 10000}
it seems to work!
I do not understand why with version 1.1.8 everything worked with higher values.

thinkyhead, 2018-09-09:
The planner and stepper portions of Marlin received a major overhaul for 1.1.9, so it's possible that bugs in 1.1.8 that caused higher values to be over-constrained were fixed.

closed 2018-09-09 by thinkyhead