MarlinFirmware/Marlin #11744: Help with UBL

eried, 2018-09-06:
Hi, I need a small hint to get my machine ready. I used to have a 3 point grid auto-level routine per print (`G28 G29`) but I upgraded and want to use UBL. I did the first run by heating the printer, doing the cold mesh generation, saving the mesh to slot 1. My startup code has:
```
G28
...
G29 L1
G29 J
```

But my prints are horrible. The nozzle goes up and down in like a zigzag. So, my question is basically: 
Is that startup code the right way to work with UBL? (`G29 J` goes too far back, to an area where the UBL mesh generation never probed)

Roxy-3D, 2018-09-06:
Two suggestions...  First set your 3 point probe points to be on the bed and where the Z Probe can reach it:
```
  #define PROBE_PT_1_X 35
  #define PROBE_PT_1_Y 365
  #define PROBE_PT_2_X 35
  #define PROBE_PT_2_Y 35
  #define PROBE_PT_3_X 365
  #define PROBE_PT_3_Y 35
```

The other suggestion is get your mesh well defined.  Use G26 to see how well it reflects the topology of your bed and use G29 P4 to edit any mesh point(s) that are not perfectly correct.   When you have your mesh well defined, you shouldn't need the 3 point tilting of the mesh.

Roxy-3D, 2018-09-06:
Yes.   But I don't use that means to get to things.  I always use PronterFace.    

Here is an example of a very good looking G26 pattern:
https://github.com/MarlinFirmware/Marlin/issues/7021#issuecomment-315636600

And here is UBL being used to print on an Easter Egg.  Your bed is not going to be as warped as that!
https://github.com/MarlinFirmware/Marlin/issues/10218

I honestly believe you won't choose to use 3-point leveling after you have a well tuned mesh!

eried, 2018-09-06:
Great! Just to confirm, that means just using:

```
G28
...
G29 L1
```

for the startup routine?

Just for completion, I'll use
`#define PROBE_PT_3_Y Y_MAX_BED-(MIN_PROBE_EDGE*2)`

But maybe it wont be used 👍

Roxy-3D, 2018-09-06:
I edited your startup GCode.    (You want the mesh loaded...  but you also want to Activate UBL)
It is possible to have UBL active (with the specified mesh loaded) when the printer powers up...  But for now...  Putting that in your slicer will work fine.

eried, 2018-09-07:
OK! it works nicely. I am going to make a quick video when I master more stuff about UBL, but feels so sweet. 

```
Bed Topography Report:

    (  1,269)                                                                                                      (209,269)
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14      
14 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .    
   |
13 |   .     +0.759  +0.855  +0.934  +0.992  +1.042  +1.054  +1.072  +1.068  +1.050  +1.029  +0.986  +0.920  +0.845    .    
   |
12 |   .     +0.736  +0.835  +0.918  +0.964  +1.005  +1.040  +1.060  +1.095  +1.063  +1.042  +0.989  +0.924  +0.838    .    
   |
11 |   .     +0.734  +0.833  +0.907  +0.969  +1.001  +1.032  +1.065  +1.066  +1.055  +1.018  +0.973  +0.904  +0.846    .    
   |
10 |   .     +0.723  +0.820  +0.898  +0.959  +1.008  +1.036  +1.069  +1.054  +1.042  +1.010  +0.978  +0.933  +0.867    .    
   |
 9 |   .     +0.684  +0.784  +0.873  +0.930  +0.976  +1.003  +1.029  +1.029  +1.025  +0.986  +0.941  +0.889  +0.815    .    
   |
 8 |   .     +0.610  +0.718  +0.813  +0.868  +0.920  +0.965  +0.966  +0.973  +0.955  +0.931  +0.893  +0.829  +0.741    .    
   |
 7 |   .     +0.560  +0.666  +0.743  +0.757  +0.774  +0.827  +0.871  +0.882  +0.875  +0.867  +0.816  +0.760  +0.681    .    
   |
 6 |   .     +0.471  +0.574  +0.652  +0.681  +0.682  +0.734  +0.764  +0.786  +0.782  +0.755  +0.718  +0.656  +0.586    .    
   |
 5 |   .     +0.365  +0.469  +0.549  +0.596  +0.638  +0.665  +0.680  +0.680  +0.655  +0.629  +0.602  +0.539  +0.480    .    
   |
 4 |   .     +0.240  +0.340  +0.415  +0.461  +0.513  +0.540  +0.540  +0.547  +0.521  +0.496  +0.461  +0.405  +0.333    .    
   |
 3 |   .     +0.091  +0.186  +0.254  +0.294  +0.335  +0.364  +0.375  +0.375  +0.356  +0.318  +0.269  +0.224  +0.150    .    
   |
 2 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .    
   |
 1 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .    
   |
 0 |[  .   ]   .       .       .       .       .       .       .       .       .       .       .       .       .       .    
        0       1       2       3       4       5       6       7       8       9      10      11      12      13      14      
    (  1,  1)                                                                                                      (209,  1)

```

The G26 pattern is really crappy on the boundaries but my sensor offset is big so it was expected. 

[image]

Weird that the bend is there, because I have 1 screw in the left and 2 on the right. This is the best bed from 3 I ordered (I was trying to eventually get a flat one...)

closed 2018-09-07 by eried

eried, 2018-09-26:
My prints are perfect now! UBL is amazing! thanks ;)

I wrote some instructions on how to. It was a bit confusing in the beginning. I ended using a excel spreadsheet to aid with the calibration process.

[image]

thinkyhead, 2018-09-27:
Your writeup at https://github.com/eried/Research/tree/master/RepRap/_UBL is pretty good. May we convert it to Jekyll markdown and post it as a page on the http://marlinfw.org website?