MarlinFirmware/Marlin #11748: [Request] Alternative to Z_AFTER_PROBING after homing, regardless of probing

1icri, 2018-09-06:
### Description
Notes:
1. I may have missed an obvious feature that does this already, please don't hit me too hard if I did.
2. Any time I mention homing I'm referring to any homing command that includes the Z axis.

I would like to propose that Z_AFTER_PROBING should be at least partially replaced with a Z position to apply immediately after homing (propsed Z_AFTER_HOMING) regardless of the homing/probing method. Currently, the only way to apply a Z motion immediately after homing is by using a probe (which many users don't have), or ensuring a suitable move command is issued after every home (which users may not want to do or bother to check).

As has been discussed several times on here, there are issues with MBL and downward-warped beds on machines with mechanical limit switches, where various solutions risk either crashing the nozzle into the bed (including sideways) or crushing and potentially damaging a mechanical limit switch.
Here are some example cases:
1. The limit switch is level with the highest point on the bed. Any downward correction crushes the limit switch.
2. The limit switch is level with the lowest point on the bed. A poorly planned move command (not necessarily manual) can crash the nozzle into a high point of the bed in several ways, and the manual leveling process itself can crash the nozzle into the bed.

By adding an option to force a Z move after homing, a solution to such cases becomes fairly trivial.
As prerequisites, MIN_SOFTWARE_ENDSTOPS (or _Z) needs to be enabled, and LCD_PROBE_Z_RANGE must be set appropriately if the user intends to use LCD bed leveling.
The home position is set outside the software end stop, the limit switch is set below the lowest point of the bed and Z_MIN_POS is set to just above the highest point of the bed. Finally, immediately after homing, Z is forced above the bed. Now, the limit switch can't be crushed at even the lowest point of the bed, and the only way the nozzle can crash into the bed is by manually driving it into the bed during MBL setup.

While I can't think of any reason right now, it may also be beneficial to force a particular X and Y after homing too, so a better solution based on this principle may be to implement settings for all three axes, or simply have a setting (or per-axis settings) to force a move inside any enabled software endstops (which may be more intuitive).

marcio-ao, 2018-10-10:
@1icri : In LulzBot firmware, we have implemented a feature that backs away from the endstops after homing. We did this because with TMC sensorless homing, we found that leaving the axis parked on endstops is a sure way to get skipped steps if another home attempt is made directly after the first. Moving away from the endstops eliminates that possibility.

I wonder if having this ability to move a certain number of mm away from endstop switches on X, Y, or Z after homing would be a more generic way of accomplishing what you are asking for?

1icri, 2018-10-11:
@marcio-ao that would also work, yes. It would mean a bit of more configuration is necessary but would open up the possible uses.

I have realised though, if MBL operates properly outside the bed area then this might not be an issue. I'll test further when my printer is working again.

lutorm, 2019-03-31:
I'm running a custom Marlin on my TAZ6 and I'm in exactly this situation. The TAZ6 Z-homing leaves the nozzle below the z-button perimeter so if you forget to raise it after probing you crash into the button steel perimeter.

@marcio-ao is the backaway feature you mention in the lulzbot marlin that you can download?

marcio-ao, 2019-04-01:
@lutorm: Take a look at our "devel" branch of Marlin. It has those modifications in place. If you want to see how it differs from stock Marlin, run "git difftool devel upstream_2.0.0 -- Marlin" from the repo checkout.

comps, 2019-04-02:
When I was making a vanilla Marlin build for my TAZ6, I also got surprised how Lulzbot decided to solve it. It was some pretty obscure code (to me):
```
+    #define LULZBOT_BACKOFF_AFTER_HOME \
+        { \
+            constexpr int x = LULZBOT_BACKOFF_X_POS; \
+            constexpr int y = LULZBOT_BACKOFF_Y_POS; \
+            constexpr int z = LULZBOT_BACKOFF_Z_POS; \
+            const bool saved_endstop_state = Endstops::global_enabled(); \
+            Endstops::enable_globally(false); \
+            do_blocking_move_to_z  ((home_all || homeZ) ? z : current_position[Z_AXIS], LULZBOT_BACKOFF_FEEDRATE); \
+            do_blocking_move_to_xy ((home_all || homeX) ? x : current_position[X_AXIS], \
+                                    (home_all || homeY) ? y : current_position[Y_AXIS], LULZBOT_BACKOFF_FEEDRATE); \
+            Endstops::enable_globally(saved_endstop_state); \
+        }
```

So instead, I just did
```diff
diff --git a/Marlin/src/gcode/calibrate/G28.cpp b/Marlin/src/gcode/calibrate/G28.cpp
index 89884f87f..038ef0901 100644
--- a/Marlin/src/gcode/calibrate/G28.cpp
+++ b/Marlin/src/gcode/calibrate/G28.cpp
@@ -151,6 +151,9 @@
 
       do_blocking_move_to_xy(destination[X_AXIS], destination[Y_AXIS]);
       homeaxis(Z_AXIS);
+
+      // Go up from TAZ6 home button (DualExtruderV3)
+      enqueue_and_echo_commands_P(PSTR("M420 S0\nG21\nG90\nG0 Z2\nG0 X40"));
     }
     else {
       LCD_MESSAGEPGM(MSG_ZPROBE_OUT);
```
and it works for me, though having a dedicated feature would work better. FTR, just the Z move is enough, ignore the mess around it.

marcio-ao, 2019-04-02:
> When I was making a vanilla Marlin build for my TAZ6, I also got surprised how Lulzbot decided to solve it.

@comps: Yeah, its kind of messy, which is why I never pushed it upstream...

But be aware of a possible flaw in your implementation. `enqueue_and_echo_commands_P` puts it at the end of the command queue, which could already contain queued moves that follow the `G28` command. So in those situations, your head could still crash.  One simple solution would be to use `GcodeSuite::process_subcommands_now` which executes the commands immediately.

Your script also uses G21 and G90 and does not save and restore the previous state, which could possibly be something that the slicer was not expecting, so it's not a general solution.

marcio-ao, 2019-04-02:
Anyhow, just to explain our implementation. It saves the endstop state and the "backs off" from the endstops by the distances LULZBOT_BACKOFF_X_POS, LULZBOT_BACKOFF_Y_POS and LULZBOT_BACKOFF_Z_POS.

In the case of the TAZ, only LULZBOT_BACKOFF_Z_POS is used, but on our Mini 2, we back off on all axis. This is because we use sensorless homing on those machines and leaving the axis against the bump stops can cause chattering if the machine is re-homed immediately afterwards.

So implementing a general feature for moving away from the endstops after hominh in all axis is preferable to implementing a feature for just raising Z.

marcio-ao, 2019-04-02:
If Marlin just had a ENDSTOP_BACKOFF_DIST_X, ENDSTOP_BACKOFF_DIST_Y and ENDSTOP_BACKOFF_DIST_Z, that would be nice...

marcio-ao, 2019-04-02:
@comps: I made a cleaned up version of our code and submitted it as a pull request for Marlin (PR#13557)

comps, 2019-04-02:
@marcio-ao Cool, thanks! I figured your code was non-trivial exactly because of X/Y backoff, but I needed the smallest, simplest, least-conflicting and easiest-to-rebase hack I could make, not a "proper" solution.
I spent a while reading `queue.cpp` and surrounding code and came to the conclusion that it would not, in fact, queue it after any other commands - and it seems I was right when the G28 was executed "inside" MLB's G29, the homing move and my commands were executed before the other code in `G29.cpp` or any other commands following G29.
Either way, it's an ugly hack, I agree.

G21 and G90 are modal, but any self-respecting CAM or slicer will issue those right before G1/G2/G3, same with feedrate. I know slicers like to break CNC traditions, but they positively surprised me here. :)

If/when that pull request gets merged, I'd even suggest leaving it on as default, similarly to how Marlin has Z_HOMING_HEIGHT enabled by default (which might be a bug, looking at how it actually gets enabled via conditionals).

boelle, 2019-06-22:
wrong label

InsanityAutomation, 2019-06-22:
This is actually done and can be closed.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this is done too

closed 2019-10-15 by boelle