MarlinFirmware/Marlin #11751: cant compile marlin 1.1.9

jon012198, 2018-09-07:
i did change the filament runout pins but  that is it  using a mks base  here is my config this sensor uses the x max endstop
[Marlin-1.1.9 jon main.zip](https://github.com/MarlinFirmware/Marlin/files/2359126/Marlin-1.1.9.jon.main.zip)

 exit status 1
pasting "/* Digital 3  / Pin 5   / PE3  / SERVO2_PIN*/" and "_DDR" does not give a valid preprocessing token

brandstaetter, 2018-09-07:
Duplicate of #10862

jon012198, 2018-09-07:
here is the error i am getting

<details>
<summary>error log </summary>
<br>
Arduino: 1.8.6 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lop12\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10806 -build-path C:\Users\lop12\AppData\Local\Temp\arduino_build_702550 -warnings=none -build-cache C:\Users\lop12\AppData\Local\Temp\arduino_cache_272396 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\lop12\Documents\marlin\Marlin-1.1.9 jon main\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lop12\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10806 -build-path C:\Users\lop12\AppData\Local\Temp\arduino_build_702550 -warnings=none -build-cache C:\Users\lop12\AppData\Local\Temp\arduino_cache_272396 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\lop12\Documents\marlin\Marlin-1.1.9 jon main\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\lop12\\AppData\\Local\\Temp\\arduino_build_702550\\sketch\\Marlin.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\lop12\\AppData\\Local\\Temp\\arduino_build_702550\\sketch\\G26_Mesh_Validation_Tool.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\lop12\\AppData\\Local\\Temp\\arduino_build_702550\\sketch\\I2CPositionEncoder.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\lop12\\AppData\\Local\\Temp\\arduino_build_702550\\sketch\\M100_Free_Mem_Chk.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\lop12\\AppData\\Local\\Temp\\arduino_build_702550\\sketch\\MarlinSerial.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\lop12\\AppDa [... cut ...]
C:\Users\lop12\AppData\Local\Temp\arduino_build_702550\sketch\stepper.cpp:2513:25: note: in expansion of macro 'READ'

       SERIAL_PROTOCOLLN(READ(E1_MS2_PIN));

                         ^

C:\Users\lop12\AppData\Local\Temp\arduino_build_702550\sketch\stepper.cpp:2513:30: note: in expansion of macro 'E1_MS2_PIN'

       SERIAL_PROTOCOLLN(READ(E1_MS2_PIN));

                              ^

exit status 1
pasting "/* Digital 3  / Pin 5   / PE3  / SERVO2_PIN*/" and "_DDR" does not give a valid preprocessing token
<details>

jon012198, 2018-09-07:
> Still a duplicate. Please see in the linked ticket how to solve this issue
> (tl;dr: downgrade Arduino IDE or remove the comments on these lines)

I did downgrade where do I  remove the comments don’t know where they are

Phr3d13, 2018-09-07:
If you downgraded Arduino IDE, you don't need to remove comments.

jon012198, 2018-09-07:
What one should I downgrade too I was useing 1.8.4

thinkyhead, 2018-09-07:
The comments that are confusing the buggy compiler are in the `pins_MYBOARD.h` file for your selected `MOTHERBOARD`. Remove end-of-line comments from all pins, or just those that the compiler chokes on.

closed 2018-09-07 by thinkyhead

Deneteus, 2018-09-09:
I had the same error but I am using 1.8.6 and when I went to the 
[image]

The problem is I don't see any end of line comments to remove on any of the pins.
[image]