MarlinFirmware/Marlin #11754: Bilinear leveling issues, bad G-code? [Marlin 1.1.9]

Faagerholm, 2018-09-07:
Hello,

I have now been sitting with my Anet E10 printer for a week trying to wrap my head around this problem.

### Description

#### printer
Anet E10 stock.
upgraded with a few parts but nothing major.

I have activated AUTO_BED_LEVELING_BILINEAR and got decent parameters to work with.
My problem is not that it won't level correctly but the second I activate bed level compensation everything falls apart.

My printer works if  `M420 S1` is not in my start G-code (except the bed is not level) :) 

This strange problem is best described with a picture.

[image]

I have gone through configuration.h file line by line trying to solve my issue but cannot grasp what it could be.

### additional files and information

I have only changed my configuration.h file, 
enabled PROBE_MANUALLY
enabled AUTO_BED_LEVELING_BILINEAR
and changed to my default steps/mm.
also I have huge clamps on my printer now so have decreased the bed size to 220x250 (I know there are better ways to do this)

[Configuration.h file](https://github.com/MarlinFirmware/Marlin/files/2360291/Configuration.zip)

### Question

Have any of you experienced similar issues?

Faagerholm, 2018-09-07:
Additional information:
```bash
start
echo: External Reset
Marlin 1.1.9

echo: Last Updated: 2018-08-01 | Author: (Bob Kuhn, Anet config)
echo:Compiled: Sep  7 2018
echo: Free Memory: 11598  PlannerBufferBytes: 1232
echo:V55 stored settings retrieved (683 bytes; crc 9490)
echo:  G21    ; (mm)
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E97.00
echo:Maximum feedrates (units/s):
echo:  M203 X400.00 Y400.00 Z8.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1500 Y1500 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P300.00 R1000.00 T1000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<m                                     ax_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
```

AnHardt, 2018-09-07:
> My problem is not that it won't level correctly but the second I activate bed level compensation everything falls apart.

So nothing is leveled. It just probed, but nothing is done with that data.
What you see on the bed is, (as expected) an unleveled bed.

Please describe "falls apart".

Faagerholm, 2018-09-07:
the print should start by printing a 2 parameter skirt, If you look closely at the picture in my original post you see it prints the first half of the skirt just fine..

Now when I do an analysis of the picture I see that it looks like my printer is doing great when printing in a negative direction ( going from positive X and Y position to a negative position)
but printing in a positive direction it doesn't travel the desired length.

Faagerholm, 2018-09-07:
Just noticed some people doesn't see the picture..

Here is a link to it: https://drive.google.com/file/d/1QSS4ymCDazPZJ5pAeuV9RWf3_XbOmAkV/view?usp=drivesdk

Faagerholm, 2018-09-10:
running MBL on Repetier-Host and problems are solved

closed 2018-09-10 by Faagerholm

thinkyhead, 2018-09-10:
Generally speaking, ABL Bilinear with Manual Probe does the same thing as Mesh Bed Leveling, so either one is fine. MBL costs less program memory, so it is preferred for 128K boards.