MarlinFirmware/Marlin #11757: Error:EEPROM datasize error.

alexxy, 2018-09-07:
I get this error after updating firmware on my printer to latest bugfix-2.0.x
Setup: mks gen 1.4 board and tmc2100 drivers. Extruder is dondolo

m500 adn m502 gives same error

GMagician, 2018-09-07:
Sequence is m502 and m500 by I think you already know.

Try compiling with EEPROM_CHITCHAT then we can understand if error is coming from configuration or bed level

alexxy, 2018-09-07:
M502 works, but when i try to save it with M500 i get same error

Ok. Will test it tommorow

alexxy, 2018-09-07:
I checked my config. EEPROM_CHITCHAT was enabled.

thinkyhead, 2018-09-07:
@alexxy — Please uncomment the following option in `configuration_store.cpp`, re-flash, and then see what kind of extra error messages you get when using `M500` and `M501`. This will help us to narrow down the discrepancy.
```cpp
// Check the integrity of data offsets.
// Can be disabled for production build.
//#define DEBUG_EEPROM_READWRITE
```

GMagician, 2018-09-08:
@alexxy remember to report host echo messages when you execute m500 (this is where chitchat may helps)

alexxy, 2018-09-08:
Hi!

```
Marlin bugfix-2.0.x
Last Updated: 2018-01-20 | Author: alexxy
Compiled: Sep  7 2018
Free Memory: 2729  PlannerBufferBytes: 1488
Error:Field lcd_preheat_hotend_temp mismatch.
ttyUSB0 PC> Error:Field lcd_preheat_hotend_temp mismatch.

Error:EEPROM datasize error.
ttyUSB0 PC> Error:EEPROM datasize error.

Index: 675 Size: 683
Hardcoded Default Settings Loaded
ttyUSB0 PC> 
ttyUSB0 PC> 
ttyUSB0 PC> m502
SENDING:M502
Hardcoded Default Settings Loaded
ttyUSB0 23°> m500
SENDING:M500
Error:Field lcd_preheat_hotend_temp mismatch.
ttyUSB0 21°> Error:Field lcd_preheat_hotend_temp mismatch.
```

thinkyhead, 2018-09-09:
Try this… Replace all occurrences of `MAX_SERVOS` with `4` in `configuration_store.cpp` and see if it makes the issue go away (after at least one `M502`, `M500`).

Also, please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop that ZIP file onto your next reply so that we can use your configs for testing.

alexxy, 2018-09-09:
I have MAX_SERVOS set to 2  (one is dondolo, second is bltouch)

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2364387/configs.zip)

bill-orange, 2018-09-09:
Take a look at closed issue #11699. I had the same problem. Oddly, my attempts to gain debug information somehow seemed to fix the problem. I never got to the bottom of it.

thinkyhead, 2018-09-10:
> I have `MAX_SERVOS` set to 2 (one is dondolo, second is bltouch)

Awesome. The problem is, that macro doesn't always get propagated correctly, which is why I asked you to try replacing it with a constant `4` in `configuration_store.cpp`.

**_Also, that value should NOT be overridden in your configuration._**
You should only override `NUM_SERVOS`.

alexxy, 2018-09-10:
@thinkyhead 

it was misstype =) NUM_SERVOS was set to 2. I didnt touch MAX_SERVOS

joba-1, 2018-10-03:
I see this too. 

I have an Anycubic Kossel Pulley, that I just updated from the original firmware to bugfix-2.0.x (encouraged by some promising commits of you, brandstaetter mentioning the pulley :)
This is the diff of Configuration_h from the Kossel examples (not much...):
```
--- src/config/examples/delta/Anycubic/Kossel/Configuration.h   2018-10-03 18:35:14.987651932 +0200
+++ Configuration.h     2018-10-03 20:05:58.984776672 +0200
@@ -29 +29 @@
-#define ANYCUBIC_PROBE_VERSION 0
+#define ANYCUBIC_PROBE_VERSION 1
@@ -34 +34 @@
-#define ANYCUBIC_KOSSEL_ENABLE_BED 0
+#define ANYCUBIC_KOSSEL_ENABLE_BED 1
@@ -90 +90 @@
-#define STRING_CONFIG_H_AUTHOR "@brandstaetter, @grbd" // Who made the changes.
+#define STRING_CONFIG_H_AUTHOR "@brandstaetter, @grbd, @joba1" // Who made the changes.
@@ -154 +154 @@
-#define CUSTOM_MACHINE_NAME "ANYCUBIC Kossel"
+#define CUSTOM_MACHINE_NAME "Kossel joba1"
@@ -1255 +1255 @@
-  #define GRID_MAX_POINTS_X 9
+  #define GRID_MAX_POINTS_X 7
@@ -1629 +1629 @@
-#define LCD_LANGUAGE en
+#define LCD_LANGUAGE de
@@ -1727 +1727 @@
-//#define REVERSE_MENU_DIRECTION
+#define REVERSE_MENU_DIRECTION

```
This is the serial output resulting of selecting "Save Config" (first line) and then "Load Config" from the lcd menu.

```
echo:Settings Stored (775 bytes; crc 52034)                                                                                                      
Error:EEPROM datasize error.                                                                                                                     
echo:Index: 769 Size: 775                                                                                                                        
echo:Hardcoded Default Settings Loaded                                                                                                           
```

Maybe I misunderstood, but it does not go away with typing M502;M500<enter> in the serial console.
The response is 
`echo:Hardcoded Default Settings Loaded`
A following "Load Config" yields the same result as above last 3 lines.

I guess its not making much sense to test z-offset or other calibrations (I assume this is stored in EEPROM) while this does not work, right?
At least I experienced the delta auto calibration always failed after acknowledging the second prompt (to remove the probe).

I'll go back to 1.1.9 now, to see how far I get there and if all else fails, go back to the original fw.
But now that I have identified my board/Config and how to build this with pio: if I can help by testing something, I'll try.

brandstaetter, 2018-10-03:
Please try `Configuration > Advanced configuration > Initialize EEPROM` this should solve your issue

joba-1, 2018-10-03:
Hi, thanks for the fast response! Unfortunately I have chosen german language. I guess, you mean "Configuration/Advanced Settings/Werkseinstellungen". I did that first thing after flashing. And retried just now. No effect :(  I'll change language to en and repeat, just in case...

brandstaetter, 2018-10-03:
Ja, ich nutze meinen nur Englisch. Wenn wirklich durch deine minimalen Änderungen hier ein Problem ist müssen wir uns das noch mal genauer ansehen

En: might be caused by the language? Other changes in diff look trivial.

joba-1, 2018-10-03:
ok, using english now. Unfortunately it does not help.

I did flash, EEPROM reset, then delta config save and load.

Each of the three steps is answered by a beep (dont know if this is ok or an error indicator)

This is the full serial log of the first boot after the flash with en config:

```
start
echo: External Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2018-01-20 | Author: @brandstaetter, @grbd, @joba1
echo:Compiled: Oct  3 2018
echo: Free Memory: 3042  PlannerBufferBytes: 1504
Error:EEPROM datasize error.
echo:Index: 769 Size: 775
echo:Hardcoded Default Settings Loaded
echo:Hardcoded Default Settings Loaded
echo:Settings Stored (775 bytes; crc 52034)
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z80.00 E96.00
echo:Maximum feedrates (units/s):
echo:  M203 X100.00 Y100.00 Z100.00 E100.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z3000 E3000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000 S0.00 T0.00 X5.00 Y5.00 Z5.00 E5.00
echo:Auto Bed Leveling:
echo:  M420 S0
echo:Endstop adjustment:
echo:  M666 X0.00 Y0.00 Z0.00
echo:Delta settings: L<diagonal_rod> R<radius> H<height> S<segments_per_s> B<calibration radius> XYZ<tower angle corrections>
echo:  M665 L218.00 R97.00 H320.00 S80.00 B75.00 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H190 B60 F255
echo:  M145 S1 H240 B100 F255
echo:PID settings:
echo:  M301 P22.36 I1.63 D76.48
echo:  M304 P374.03 I72.47 D482.59
echo:Z-Probe Offset (mm): (mm):
echo:  M851 Z-19.00
echo:Filament load/unload lengths:
echo:  M603 L650.00 U750.00
echo:Settings Stored (775 bytes; crc 52034)
Error:EEPROM datasize error.
echo:Index: 769 Size: 775
echo:Hardcoded Default Settings Loaded

```

joba-1, 2018-10-03:
I flashed 1.1.9 now and M502, M500, M501 series works fine (the last lists a bunch of settings on the serial console. Init EEPROM and save/load settings via menu works as well. So I guess it is not a hardware problem.

closed 2018-10-04 by thinkyhead