MarlinFirmware/Marlin #11760: Why my printer is ignoring the endstops?

Xenoyr, 2018-09-07:
Hi, first of all sorry for my bad english, and thanks
I open another thread because the old oine has been closed
But anyway I have the same problem, My printer is ignoring the endstop
If the logic is setted true/false, or the endstop is constantly active/closed
so, or the mottors can go to only one direction, segnalating that the endstop is constatly active
or if they can go to both direction, but the endstop arent readen
so, isnt a problem of the  logic of endstop, and also it isnt a problem with bad connection of cables

Im using the more recent version of Marlin
with an trigorilla 2560 1.4 and ctc i3 wood body
with tmc2208 driver

And Also I follow this scheme
[image]

GMagician, 2018-09-07:
Have you tested your endstops with M119?
No press should be all "open" and when you keep pressed one, it should give you "triggered".
Post also your current configuration (attach files)

Xenoyr, 2018-09-09:
@GMagician the z and y are showed as triggered
for the files.. after because.. well I change my pc
and for any reason, transferring the files
even with copy and pasting (idk)
my changes of marlin are gone, so actually, I dont have a copy of my configuratione >.<

brandstaetter, 2018-09-10:
Check out #11730 for a configuration example for the Anycubic Kossel - you can use it as starting point to modify things to your needs

Xenoyr, 2018-09-11:
@GMagician Im using this configuaration:
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2373030/marlin.zip)

Xenoyr, 2018-09-14:
up

GMagician, 2018-09-14:
@Xenoyr it's ok that you have open when micros are released, but when you press them? M119 reports proper endstop triggered?

Xenoyr, 2018-09-14:
@GMagician Even when I press them, they say Always open
but if I reverse the logic of endstop, opposite problem
always triggered

AnHardt, 2018-09-14:
Do you have the right board configured?
Test the boards pins. Take a wire and connect the pins to ground -- M119. Then connect to 5 or 3.3V - M119. If neither shows triggered you are either testing the wrong pins, or they are defect.
Test the endstops without the board, with just power and a voltmeter.
If the switches do work, connect them correctly.

Xenoyr, 2018-09-14:
@AnHardt the endstops works..  I just tested them in a different board
yeah.. I change the board name, and take the name from the board section
Im a newbie.. so I think I didnt understanstand well what I need to do for testing the pinboards..

brandstaetter, 2018-09-15:
Which board hardware are you using, maybe you picked the wrong board name?

Xenoyr, 2018-09-15:
@brandstaetter trigorilla 2560 1.4,

Xenoyr, 2018-09-23:
up

Xenoyr, 2018-09-23:
I dont think its the motherboard, because I just buy another 3d printer (anycubic i3 mega), (but nevere tryed with the basics driver), and with tmc2208 I still have the same problem

Xenoyr, 2018-09-23:
@brandstaetter , @AnHardt , @GMagician  maybe I undestand why, the steppers can work in a determinate area, if this dont coincide with the actually aerea of the 3d printer, they can go over the actually area of the 3d printer, even ignoring the endstop, and also if the area of the steppers < 3d printer aerea, they also dont touch any "barrier" of the 3d printer, how I can solve this issue?

GMagician, 2018-09-23:
ENDSTOPS_ALWAYS_ON_DEFAULT will keep endstops always under control. But if your software limits (the ones in config) are smaller...no way to move outside them

Xenoyr, 2018-09-23:
@GMagician Thanks, I watch the configuration that I was using, and also a marlin stock one, that was commented; but what if the area doesnt coincide with the 3d printer? anything like this:  https://imgur.com/a/nCt2xgO

boelle, 2019-02-20:
@Xenoyr problem solved?

closed 2019-02-20 by thinkyhead