MarlinFirmware/Marlin #11763: makerbase 3d touch problem on my tarantula makerbase gen l 1.0

tsiras, 2018-09-08:
hi to everybody
i feel like a nerd or  ....  stiubit 

### Description
i try to make the tt (tevo tarantula mks gen l 1.0 with the mks 3dtouch ) i install everything i plug us it explained at alexp page all the modifications for autobed leveling 
but.... nothing

i use all the instructions but nothing the (bl touch clone i think is control the pin is do the touch check i give all the commands the pin is going in and out  but the zmin gnd to sing dont feels anything)
[image]
 .  i connect the z pin cable at gnd  and d18 (zmin)  and  servo cables d11 gnd 5v

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2363107/Marlin.zip)
[instructions.docx](https://github.com/MarlinFirmware/Marlin/files/2363109/instructions.docx)
i feel so ........ bad   i check the gnd and d18 pins unplugged the cables  have 16,7Ω STOW AND 20,4Ω DEPLOY
please if anyone have any idea please help

tsiras, 2018-09-08:
THE SERVO CONNECTION IS  D11 5V GND WITH THIS PINING

AnHardt, 2018-09-08:
The BL-Touch and clones give a 5-10ms trigger **pulse**. Unlikely to see this on a voltmeter or with M119.

tsiras, 2018-09-08:
i understand that but when i run the bed leveling the pin is puled up but the z moving never stops like the zmin is not feel the change of statement .all the functions are work but the auto bed leveling is try to brake my tarantulas plastic parts......i read in a forum that the min & max pins have a  1Kohm resistor each in serial i m waiting the maker base ltd answer about my problem because im not sure and
i dont have any idea about this ....i ask the makerbase ltd about this malfunction !!!! if you have time i have all the info and the config files i change and use for this...... the link of the 3dtouch of makerbase too
best regards

boelle, 2019-02-20:
@tsiras problem solved?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead