MarlinFirmware/Marlin #11764: Can't compile Marlin 2 when enabling Bltouch

julianCast, 2018-09-08:
### Description

### Steps to Reproduce

1. Verify Marlin 2.x arduino 1.9.0-beta on **enabling bltouch** and BILENEAR

**Expected behavior:** Verification valid.

**Actual behavior:** It shows an error:
`In file included from sketch/src/HAL/shared/servo.cpp:60:0:
sketch/src/HAL/shared/servo_private.h:64:0: warning: "MAX_SERVOS" redefined
 #define MAX_SERVOS   (_Nbr_16timers  * SERVOS_PER_TIMER)
 ^
In file included from sketch/src/HAL/shared/../../inc/MarlinConfig.h:30:0,
                 from sketch/src/HAL/shared/servo.cpp:54:
sketch/src/HAL/shared/../../inc/../pins/pins.h:629:0: note: this is the location of the previous definition
   #define MAX_SERVOS 4
 ^
In file included from sketch/src/HAL/HAL_AVR/servo_AVR.cpp:64:0:
sketch/src/HAL/HAL_AVR/../shared/servo_private.h:64:0: warning: "MAX_SERVOS" redefined
 #define MAX_SERVOS   (_Nbr_16timers  * SERVOS_PER_TIMER)
 ^
In file included from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:30:0,
                 from sketch/src/HAL/HAL_AVR/servo_AVR.cpp:56:
sketch/src/HAL/HAL_AVR/../../inc/../pins/pins.h:629:0: note: this is the location of the previous definition
   #define MAX_SERVOS 4
 ^
Sketch uses 161478 bytes (63%) of program storage space. Maximum is 253952 bytes.
Global variables use 4916 bytes (60%) of dynamic memory, leaving 3276 bytes for local variables. Maximum is 8192 bytes.`

[confs.zip](https://github.com/MarlinFirmware/Marlin/files/2363114/confs.zip)

#### Additional Information
[image]

julianCast, 2018-09-08:
I didn't give it a try, I was afraid I was going to upload something uncompleted, should I?

Atatoth, 2018-09-08:
Compile in IDE 1.8.5

fiveangle, 2018-09-09:
With your configs...

```
avr-objcopy -O ihex -R .eeprom .pioenvs/megaatmega2560/firmware.elf .pioenvs/megaatmega2560/firmware.hex
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  36.3% (used 2977 bytes from 8192 bytes)
PROGRAM: [===       ]  29.4% (used 74684 bytes from 253952 bytes)
text	   data	    bss	    dec	    hex	filename
74396	    288	   2689	  77373	  12e3d	.pioenvs/megaatmega2560/firmware.elf
========================= [SUCCESS] Took 20.03 seconds =========================

================================== [SUMMARY] ==================================
Environment megaatmega2560      	[SUCCESS]
Environment megaatmega1280      	[SKIP]
Environment at90usb1286_cdc     	[SKIP]
Environment at90usb1286_dfu     	[SKIP]
Environment DUE                 	[SKIP]
Environment DUE_USB             	[SKIP]
Environment DUE_debug           	[SKIP]
Environment LPC1768             	[SKIP]
Environment melzi               	[SKIP]
Environment melzi_optiboot      	[SKIP]
Environment rambo               	[SKIP]
Environment sanguino_atmega644p 	[SKIP]
Environment sanguino_atmega1284p	[SKIP]
Environment STM32F1             	[SKIP]
Environment STM32F4             	[SKIP]
Environment teensy35            	[SKIP]
Environment malyanm200          	[SKIP]
Environment esp32               	[SKIP]
========================= [SUCCESS] Took 20.03 seconds =========================

```

thinkyhead, 2018-09-09:
@p3p, @xC0000005, @hasenbanck — The definition of `MAX_SERVOS` is problematic. See `servo_private.h` and contrast with the other defines of `MAX_SERVOS`. The definition for AVR is also kind of odd…
```cpp
#define SERVOS_PER_TIMER       12     // the maximum number of servos controlled by one timer
#define MAX_SERVOS (_Nbr_16timers * SERVOS_PER_TIMER)
```
I'm just testing how it compiles without being defined earlier (as 4) by `pins.h`, and the above definition actually evaluates to zero.

hasenbanck, 2018-09-09:
@thinkyhead — The definition in case of the official STM32 core comes from the Servo library and is the same as your quoted lines. In my case it evaluates to 12 (_Nbr_16timers=1).

I also have a working BLtouch configuration running with the STM32 core.

Did I understand you right?

thinkyhead, 2018-09-09:
The problem is that the servos code in the HAL is kind of a mess, and there are conflicting definitions. I think we're going to have to overhaul the servos portion of the HALs so everything is more coherent.

thinkyhead, 2018-09-09:
The conflicting definitions of `MAX_SERVOS` is possibly what's causing EEPROM errors too, as with #11757. We'll know more about that soon.

hamster65, 2018-09-10:
I pulled the 2.0 branch just now and it compiled fine for Arduino Mega with bltouch. But it also shows the MAX_SERVO warnings. No issues during print and leveling (linear).

thinkyhead, 2018-09-11:
To explain what's up with the `MAX_SERVO` issue, it's basically that on AVR we have an existing limit of 4, even though the HAL defines something like 12 per timer. The limit of 4 is defined by the board, since we know how many servo connectors are on each type of board. The `MAX_SERVOS` value defined by `servo_private.h` is about what the servo class can accommodate rather than what the physical plug count is on the board. In addition, only a few select files actually include `servo_private.h` from the active HAL, so the value defined there isn't even seen by much of the code.

I believe that the `MAX_SERVOS` defined in the HAL needs a new name, such as `MAX_SERVO_OBJ` to differentiate between the size the servo class will use for its internal `servos` array versus the size of the `servo_angles` array that stores the list of connected servos Marlin is using for its own functions.

I'll try to make a coherent patch based on this difference.

hasenbanck, 2018-09-11:
@thinkyhead Are you sure that you can give the MAX_SERVOS a new name? As far as I know this macro comes from the servo library ([for example](https://github.com/stm32duino/Arduino_Core_STM32/blob/ca36657bca6baf5b7600375d9dedae98f886fd0c/libraries/Servo/src/Servo.h#L86)).

thinkyhead, 2018-09-11:
I can rename the one I invented for Marlin, while leaving the library define as it is.

julianCast, 2018-09-12:
@thinkyhead Would this (https://github.com/MarlinFirmware/Marlin/commit/8584d3697d1e0bebe67e8d30184ef7dc04d873e6) fix the problem? I can try this afternoon.

thinkyhead, 2018-09-12:
> Would this … fix the problem?

That is my hope!

julianCast, 2018-09-13:
It's compiling with no warnings! Thank you!

closed 2018-09-13 by julianCast

Mafer2018, 2018-10-02:
Hey I'm having the same problem as you with MAX_SERVO but I don't know how to fix it, I'm new at programming and I would really appreciate your help

Mafer2018, 2018-10-02:
@thinkyhead How do you change the name to MAX_SERVO_OBJ?

anm8tor, 2019-06-17:
@thinkyhead and @hasenbanck , I know this issue was closed, but I am experiencing the same issue, trying to configure an STM32 board, the SKR Mini V1.1. I started with @3DSmitty's version of Marlin for these boards, but am getting the MAX_SERVOS conflict. Any suggestions?