MarlinFirmware/Marlin #11765: Z_homing Error

Studna3D, 2018-09-08:
Hi, 
I have problem with Z_homing. When I set home all axes, home X and home Y is OK, but Z home no. 
This is my problem:  https://www.youtube.com/watch?v=FFUWMU84zdo
I use TMC2130 driver with SPI for all axis and extruder.

```
09:50:24.324 : Z driver error detected:
09:50:24.324 : short to ground (coil A)
09:50:24.324 : X Y Z E0
09:50:24.324 : Enabled  true true true false
09:50:24.324 : Set current 900 900 1000 900
09:50:24.324 : RMS current 887 887 994 887
09:50:24.324 : MAX current 1251 1251 1402 1251
09:50:24.324 : Run current 28/31 28/31 17/31 28/31
09:50:24.324 : Hold current 14/31 14/31 8/31 14/31
09:50:24.324 : CS actual  14/31 14/31 17/31 14/31
09:50:24.324 : PWM scale  20 20 0 0
09:50:24.324 : vsense  1=.18 1=.18 0=.325 1=.18
09:50:24.324 : stealthChop true true true true
09:50:24.324 : msteps  16 16 16 16
09:50:24.324 : tstep  1048575 1048575 31 1048575
09:50:24.324 : pwm
09:50:24.324 : threshold  39 39 13 277
09:50:24.324 : [mm/s]  101.36 101.36 15.20 30.04
09:50:24.324 : OT prewarn false false false false
09:50:24.324 : OT prewarn has
09:50:24.324 : been triggered false false false false
09:50:24.324 : off time  5 5 5 5
09:50:24.434 : blank time 24 24 24 24
09:50:24.434 : hysteresis
09:50:24.434 : -end  2 2 2 2
09:50:24.434 : -start  3 3 3 3
09:50:24.434 : Stallguard thrs 0 0 0 0
09:50:24.434 : DRVSTATUS X Y Z E0
09:50:24.434 : stallguard
09:50:24.434 : sg_result  0 0 0 0
09:50:24.434 : fsactive
09:50:24.434 : stst  X X  X
09:50:24.434 : olb    X
09:50:24.434 : ola
09:50:24.434 : s2gb
09:50:24.434 : s2ga    X
09:50:24.434 : otpw
09:50:24.434 : ot
09:50:24.434 : Driver registers: X = 0x80:0E:00:00
09:50:24.434 : Y = 0x80:0E:00:00
09:50:24.434 : Z = 0x48:11:00:00
09:50:24.434 : E0 = 0x80:0E:00:00
09:50:24.434 : Firmware was halted, trying to reconnect. Eventually running print is stopped.
09:50:25.364 : Error:Printer halted. kill() called!
```

thinkyhead, 2018-09-09:
As a safety feature we added a homing validation after homing, which checks to make sure that the last homing move actually resulted in an endstop being triggered. This is a safety feature that works for most setups, but if it doesn't work for yours, then for now it can be worked around by commenting out any calls to `endstops.validate_homing_move()` throughout the code.

At some point @ejtagle and I will audit the endstops code and fix any lingering quirks from the recent planner/stepper/endstops refactor.

jfint, 2018-09-09:
of course a day later I can home 100% 15 times in a row on just X and Y without a failure to get the printed error message.  I'll post when I get it, because I don't think it was "Home Failed" which is what looks like the code sends for this particular check.

jfint, 2018-09-10:
Thinkhead, I tried your suggestion and it homes a treat, but then it stars the bed probe from the center of the bed instead of from 0,0

thinkyhead, 2018-09-10:
@jfint — That would have to be a separate issue. It sounds like your endstop switches are throwing false signals. Does `ENDSTOP_NOISE_FILTER` help at all?

jfint, 2018-09-10:
@thinkyhead I think I agree,I may just bring an OScope home from work and see if I can see the issue.  I suspect that the optical sensors might not be reliable with MKS Sbase

thinkyhead, 2018-09-10:
N-C switches are preferred over N-O because they're immune to induction. You can add a resistor / capacitor to the endstop lines to clean up the signals and this usually eliminates noise that could lead to false triggers.

Studna3D, 2018-09-12:
I tried `ENDSTOP_NOISE_FILTER` but it is the same.

closed 2018-11-02 by thinkyhead