MarlinFirmware/Marlin #11771: set z home offset

marcostrullato, 2018-09-08:
Hi everyone, 

I have a custom anet a8 where I installed marlin 1.1.8. 
The printer has an hallon sensor (https://www.thingiverse.com/thing:2838259) 

I'm struggling with setting the offset, it looks like something funky is going on.

These are the command I'm sending to set home:

```
>>> g28
SENDING:G28
>>> g0 z0
SENDING:G0 Z0
echo:Probe Z Offset: -1.09
>>> M851 Z-1.09
SENDING:M851 Z-1.09
echo:Probe Z Offset: -1.09
>>> m500
SENDING:M500
echo:Settings Stored (642 bytes; crc 45841)
```

Then to verify the setting I go z home

```
>>> g0 z0
SENDING:G0 Z0
```

With this the printer moves back up 1.09 cm.

What do you think?

jfint, 2018-09-08:
I had this happen, send a G28.  Then tgry a G1 Z0, are teh G28 it worked for me.

thinkyhead, 2018-09-09:
First, update to 1.1.9.

Second, `M851` is for a probe, not for an endstop. `M206` is the command for setting a home offset.

boelle, 2019-02-20:
@marcostrullato tried what thinky suggested?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead