MarlinFirmware/Marlin #11774: Stepper Motors Sqeal with Bed PID Turned on

jfint, 2018-09-09:
### Description

With the Bed heater PID Turned on in 2.0 "bugfix" , on MKS SBase the stepper Motors squeal and do not move when commanded.  Turn off that one setting and All is well

### Steps to Reproduce

1. Enable HeatBed PID
2. Send Any Movement (G1 X50 for example)

#### Additional Information

forkoz, 2018-09-09:
I have used PID heatbed this entire time and never encountered this.

hamster65, 2018-09-09:
I might have this problem on my arduino mega with ramps 1.4. I enabled bed PID today. The printer did G28 and G29 just fine and then waited to reach the set temperature, according to my start gcode. 

When it started the actual print, at least one stepper motor made a "bad" sound and the nozzle crashed into the bed. 

This happened 3 times in sequence. Then I lowered the z acceleration and maximum feed rate and now it is printing fine again. I am sure I was not near the mechanical limit and the same acceleration and speed settings have always worked fine. It was clearly not an issue with loosing steps because the nozzle moved too far downwards.

jfint, 2018-09-09:
My checkout was from 09/06/2018, after Roxy-3D's checkin.

thinkyhead, 2018-09-10:
See if anything changes by reducing `MAX_BED_POWER` to 250.

jfint, 2018-09-11:
This was with MAX_BED_POWER to 250. and was commanded 10mm moves:
[https://drive.google.com/file/d/1ckscSkRehVF1Q2ZhBoAlMCl4qdNJ9G87/view?usp=sharing](url)

Bergerac56, 2018-09-11:
I use for the moment a MKS SBASE board in one of my printers , Heathbed PID is activated and MAX_BED_POWER 255. (FW: very last bugfix 2.0)

Till now I did not encounter this problem

jfint, 2018-09-11:
Meaning that now you see the problem?

Bergerac56, 2018-09-12:
No I don't. Just finished a serie of prints without any problem

lvent, 2018-09-26:
I can confirm the same behavior with my Anet A8, I have also tested heatbed pid on 1.1.8 and did not encounter a single issue while on 1.1.8

boelle, 2019-02-20:
@jfint problem solved?

jfint, 2019-02-20:
not that I know of, however I have noticed that I have really no need for PID control of my bed, its large and has a natural historesis that makes the bang-bang controller work fine.

boelle, 2019-02-20:
what stepper drivers do you have on that Sbase board?
can PSU supply enough current?

jfint, 2019-02-20:
PSU is 1000W, should be plenty.  I'll check on the drivers when I get home they are the stock for MKS Sbase

boelle, 2019-02-20:
some stepper drivers need 24V to be near noise free

jfint, 2019-02-20:
I run the system at 24V, steppers, heaters, etc..., save my z-axis that is 2 motors in series for max torque current, but results is 12 volts on each.

boelle, 2019-02-20:
24V each is better

jfint, 2019-02-20:
not necessarily, series vs parallel on steppers is a fun topic.  If you run at 24 volts you can often benefit from series steppers, especially if you don't want to move at the fastest possible.  Putting them in series noticeably improved their movement with the lead screws.

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead