MarlinFirmware/Marlin #11779: [2.0.x] Delta Calibration > Delta Hight: no speed up when turning knob faster

brandstaetter, 2018-09-09:
### Description

When setting the temperature, there's a speed up of the value change if you turnthe knob/encoder faster.
In the `Configuration > Delta Calibration > Delta Settings > Height` this speedup does not happen, which makes it a bit annoying if the value has to be changed by a larger amount.

### Steps to Reproduce

1. `Configuration > Delta Calibration > Delta Settings > Height` 
2. Turn encoder knob slow and fast

**Expected behavior:** bigger value change when turning faster

**Actual behavior:** same value change independent of speed

Configuration see #11730

boelle, 2019-02-20:
@brandstaetter problem solved?

brandstaetter, 2019-02-20:
@boelle nope. Did not notice any changes.

brandstaetter, 2019-02-23:
Oh, no, there were no related changes in that change.

boelle, 2019-06-22:
missing a label

thinkyhead, 2019-07-11:
Soon this will be changed. In the meantime, patch your code thusly…
```diff
- MENU_ITEM_EDIT_CALLBACK(float52sign, MSG_DELTA_HEIGHT, &delta_height, delta_height - 10, delta_height + 10, _recalc_delta_settings);
+ MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(float52sign, MSG_DELTA_HEIGHT, &delta_height, delta_height - 10, delta_height + 10, _recalc_delta_settings);
```

closed 2019-07-11 by thinkyhead