MarlinFirmware/Marlin #11782: CoreXY with TMC2208. XY Axes move at an angle.

parthaero, 2018-09-09:
### Description

### Steps to Reproduce

1. Tried Marlin (stable)/Marlin(bug fix) on MKS Gen1.4 with TMC2208
2. CoreXY system
3. Moving X and Y axes in any direction.

Expected behavior: Moving X or Y axes individually on DRV8825 or A2988 work just fine and the X and Y axes are perpendicular to each other, which should happen with TMC2208 too.

Actual behavior: But using TMC2208 the X and Y axes are not perpendicular to each other. The axes have an angle of approximately 30 degrees

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2364277/config.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-09-10:
It's possible you have damaged stepper drivers. Do you have any spares that you can test with?

parthaero, 2018-09-10:
Yes, I bought 8 of them tried replacing both X and Y axes. The same issue, moving at the same angle.

parthaero, 2018-09-10:
Also, I tried interchanging X and Y axes motor drivers. But the angle at which the axes move remain the same. If there was a problem with the driver, the angle between the X and Y axes should be opposite after interchanging X and Y axes drivers, but it does not.

parthaero, 2018-09-10:
Have found the problem

closed 2018-09-10 by parthaero