MarlinFirmware/Marlin #11783: Marlin 1.1.9 on a Anet A8

AceOfSpadezCC, 2018-09-09:
Good afternoon everyone. I am new to 3d printing, and more so to coding. I have gone through as many forums and youtube videos I could, but have yet to have any luck resolving my issue. I keep getting a #endif without #if error trying to upload the software. I purchased my A8 and a auto leveling sensor, unaware that the firmware would not have the option in it. I have been through the config file many times changing settings as told, but I keep getting the same errors telling me to enable one probe, which I did, for my fix mounted probe. Any help is greatly appreciated, I will paste the code then the error. and my config file if anyone can help.

//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

/**
 * Z_MIN_PROBE_ENDSTOP
 *
 * Enable this option for a probe connected to any pin except Z-Min.
 * (By default Marlin assumes the Z-Max endstop pin.)
 * To use a custom Z Probe pin, set Z_MIN_PROBE_PIN below.
 *
 *  - The simplest option is to use a free endstop connector.
 *  - Use 5V for powered (usually inductive) sensors.
 *
 *  - RAMPS 1.3/1.4 boards may use the 5V, GND, and Aux4->D32 pin:
 *    - For simple switches connect...
 *      - normally-closed switches to GND and D32.
 *      - normally-open switches to 5V and D32.
 *
 * WARNING: Setting the wrong pin may have unexpected and potentially
 * disastrous consequences. Use with caution and do your homework.
 *
 */
//#define Z_MIN_PROBE_ENDSTOP

/**
 * Probe Type
 *
 * Allen Key Probes, Servo Probes, Z-Sled Probes, FIX_MOUNTED_PROBE, etc.
 * Activate one of these to use Auto Bed Leveling below.
 */

/**
 * The "Manual Probe" provides a means to do "Auto" Bed Leveling without a probe.
 * Use G29 repeatedly, adjusting the Z height at each point with movement commands
 * or (with LCD_BED_LEVELING) the LCD controller.
 */
//#define PROBE_MANUALLY
//#define MANUAL_PROBE_START_Z 0.2

/**
 * A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
 *   (e.g., an inductive probe or a nozzle-based probe-switch.)
 */
#define FIX_MOUNTED_PROBE

/**
 * Z Servo Probe, such as an endstop switch on a rotating arm.
 */
//#define Z_PROBE_SERVO_NR 0   // Defaults to SERVO 0 connector.
//#define Z_SERVO_ANGLES {70,0}  // Z Servo Deploy and Stow angles

/**
 * The BLTouch probe uses a Hall effect sensor and emulates a servo.
 */
//#define BLTOUCH
//#if ENABLED(BLTOUCH)
  //#define BLTOUCH_DELAY 375   // (ms) Enable and increase if needed
#endif

[... 345 lines cut ...]
In file included from sketch\HAL.h:36:0,

                 from sketch\MarlinConfig.h:39,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:223:0: note: this is the location of the previous definition

 #define TIMER5C 18

 ^

In file included from sketch\MarlinConfig.h:42:0,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:735: error: missing binary operator before token "#"

   #error "Please enable only one probe option: PROBE_MANUALLY, FIX_MOUNTED_PROBE, BLTOUCH, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."

   ^

exit status 1
#endif without #if
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2364478/Marlin.zip)

brandstaetter, 2018-09-10:
you need to define one probe option e.g. line 636 //#define FIX_MOUNTED_PROBE

Andreaux, 2019-01-18:
There's an error here:
The BLTouch probe uses a Hall effect sensor and emulates a servo.
*/
//#define BLTOUCH
//#if ENABLED(BLTOUCH)
//#define BLTOUCH_DELAY 375 // (ms) Enable and increase if needed
#endif
/**

The #endif is uncommented, while the corresponding #if is not.
The if/endif and whatever in between is inactive as long as #define BLTOUCH is commented out anyway.
Try changing this to:

The BLTouch probe uses a Hall effect sensor and emulates a servo.
*/
//#define BLTOUCH
#if ENABLED(BLTOUCH)
#define BLTOUCH_DELAY 375 // (ms) Enable and increase if needed
#endif
/**

This might not solve all errors, but this is definitely an error.

boelle, 2019-02-20:
@AceOfSpadezCC problem solved?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

rewolff, 2019-08-13:
I have an Anet A6, and I have the same issue: a big list of TIMERxy defines are defined in two files. Because I think those are important and I don't know which is right and which is being used, I don't dare to flash my hardware with the binary produced with hundreds of warnings. 

Suggestions?

dzamosenchuk, 2019-11-12:
Hello! I have the same bugs about error. I'm using Anet a2+ configuration, Marlin 1.1.x or bugfix version and Arduino IDE 1.8.10 or 1.9 beta. And board definitions from this repo: https://github.com/SkyNet3D/anet-board .
Could You help with this issue?