MarlinFirmware/Marlin #11784: Question about building for F4/7

SJ-Innovation, 2018-09-09:
I think i'm missing something. 
Im developing an F7 powered board, thats going well, and can compile (for the most part) using the ArduinoIDE 1.9-beta with STM32Generic. I use CLion as my primary IDE, and normally use PlatformIO for programming other F4 & F7 based projects (as well as by 8-bit printers).  I am aware than PIO and the F1 boards build and run just fine, yet for the life of me I cant get STM32Generic to play ball with PIO.

Ive spent the last 3 days doing everything that seems logical to get it to work, running on the logic that "if everyone else got it to work, i should be able to too". Is everyone using F4/7 just using STM32Generic with the ArduinoIDE to upload or the moment? Or am i missing a trick? xD (If so what trick please ;P)

thinkyhead, 2018-09-10:
A good question for @xC0000005 or @hasenbanck perhaps…

hasenbanck, 2018-09-10:
Answered the question inside the pull request of my F7 HAL of the official core  (https://github.com/MarlinFirmware/Marlin/pull/11750).

SJ-Innovation, 2018-09-10:
Closing, answer in #11750. Thank you!

closed 2018-09-10 by SJ-Innovation