MarlinFirmware/Marlin #11785: Digipot clarification

ModMike, 2018-09-09:
### Description

Digipot voltage clarification

When setting digipot voltages:

1) Is the voltage being set vRef or motor current?  Assuming it's motor the way it's written
2) Are the channels referring to drivers or X,Y,Z, E0?  I assume since we have to set channels we are setting driver 1,2,3,4. etc.  If so how do they map EX: first entry = X?

What is the recommended setting for a CR-10 motor current?   1.8 (not v-ref)?

Thank you

thinkyhead, 2018-09-10:
The motor current Vref can't be set on a CR-10 using G-code unless it has been modified. There are a few good guides on YouTube about how to adjust stepper driver Vref using the physical pots on the driver. And the reprap.org website also has good explanations about the meaning of Vref and how to determine the correct setting for any random stepper motor. (Requires a multimeter.)

thinkyhead, 2018-09-12:
You will have to check the resistance on your stepper motor coils to determine the correct current setting. The reprap.org website has a short guide on how to determine the correct Vref setting for any random stepper motor.

Personally, I find it's best to just use trial-and-error until you find the lowest current that gives a good holding torque without causing the steppers to heat up when idle.

WrQth, 2019-01-08:
I have a question closely related to this except I know what the current max is the my steppers (1.68amp)... I have the Azteeg X5 Mini Wifi board and the SD2224 (TMC2224 derivative) which are digipot ready as is the Azteeg board but I am unclear when options to enable in the marlin firmware to enable setting the motor current through the software like what can be done with Smoothieware with the same setup.
2 questions...
1.) Is this possible with this setup in Marlin and if so what options need to be configured?
2.) In the firmware I see options to tell Marlin the stepper drivers but SD2224 or TMC2224 is not listed but TMC2208 which is in the same family (if I read Trinamics datasheet right) so could I define TMC2224 as the stepper driver?

@ModMike thank you letting my hijack your thread and will create an new "issue" if needed

boelle, 2019-02-20:
@ModMike so this one is still relavant?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead