MarlinFirmware/Marlin #11788: Marlin 1.9 Anet A8 SD-Card Does Not Work

helakejr, 2018-09-10:
### Description

In Marlin 1.8 Anet A8 worked fine including SD Card Support. In Marlin 1.9 SD Card inserted is not recognized.

### Steps to Reproduce

1. Build Anet A8 with example include files on Marlin 1.8 Baseline
2. Flash via arduino
3. After boot check for SD card being recognized
Works fine......

4. Build Anet A8 with example include files on Marlin 1.9 Baseline
5. Flash via arduino
6. After boot check for SD card being recognized
Fails - SD Card not recognized

**Expected behavior:** [What you expect to happen]
SD card being recognized

**Actual behavior:** [What actually happens]
SD card not being recognized.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-09-10:
Try adding this to the end of the `setup()` function in `Marlin_main.cpp`…
```cpp
#if ENABLED(SDSUPPORT)
  if (!card.cardOK) card.initsd();
#endif
```

helakejr, 2018-09-10:
That did it - I put it in Marlin_Main->Setup.

RudolphRiedel, 2018-09-29:
Just fell for this one again with my CR-10 after re-downloading 1.1.9 again and building it.
Any update on why we need to add this for 1.1.9?

jtaylor330, 2019-04-14:
Just updating to this for April 2019
Running Marlin 1.1.9 with the ANET A8 
Added the if (!card.cardOK) card.initsd();
line to the Main.cpp right after #if ENABLED(SDSUPPORT)
(As instructed ^^)
and now my SD Reads fine. 
Thanks

rotorrad, 2020-06-01:
Hi guys. I'm completely green when it comes to programming. I need to enable SD card feature on my Anet A8. I'm sure what to wrote solves the problem, I just don't know where to put it. Marlin main cp in Arduino ide. Can someone explain that to me, please.
Cheers

tm4n, 2020-06-02:
> 
> 
> Hi guys. I'm completely green when it comes to programming. I need to enable SD card feature on my Anet A8. I'm sure what to wrote solves the problem, I just don't know where to put it. Marlin main cp in Arduino ide. Can someone explain that to me, please.
> Cheers

First, this fix is for for old marlin release 1.1.9 in the no longer updated 1.1.x branch. It would be better to use the newest 2.x version instead. If you are using that new version, you have a different problem, probably a wrong configuration file.

However, to apply the fix, open the file "Marlin/Marlin_main.cpp" in the folder where you downloaded and extracted Marlin 1.1.9, with a text editor. You can also use the text editor in the Arduino SDK. 
Near the bottom of the file, you can find the line:
"void setup() {"
This is where the setup function starts. Scroll down even further from this line, until the end of the function. The end is where a line has a single closing backet "}", without any spaces in front of it.
Above that line, paste in the code. It should now look like this:
`
[...]

  #if ENABLED(SDSUPPORT)
    if (!card.cardOK) card.initsd();
  #endif
}
`
Save the changes, close the text editor, and compile and upload the marlin.ino project with Arduino, as usual.

rotorrad, 2020-06-02:
Thank you.
I decided to use 1.1.9 version as there are apparent issues between Anet board and Marlin 2.xxx
I'll give it a try.
cheers

rotorrad, 2020-06-02:
Hi again.
Is that it?
Line 15461 in Marlin main cop: 
#if ENABLED(SDSUPPORT) && !(ENABLED(ULTRA_LCD) && PIN_EXISTS(SD_DETECT))
    card.beginautostart();
  #endif
}

/**
 * The main Marlin program loop
 *
 *  - Abort SD printing if flagged
 *  - Save or log commands to SD
 *  - Process available commands (if not saving)
 *  - Call heater manager