MarlinFirmware/Marlin #11790: How to figure E2END?

boelle, 2018-09-10:
I have ordered a few Ramps FD V2.3 boards and is now trying to build up a pins file from kicad schematics

one thing that puzzles me is how to figure the E2END address, sicne the due does not have EEprom the Ramps FD have one 

its this one: https://www.mouser.com/ProductDetail/511-M24512-RMN6TP

ejtagle, 2018-09-10:
If your board has an external eeprom, E2END is just the last valid address in that eeprom chip.
If your board does NOT have an external eeprom, and you want to use the Due EEPROM emulation code, E2END should be 4095 (or 0xfff)

boelle, 2018-09-10:
board has external EEprom

how do i figure the last valid address?

ejtagle, 2018-09-10:
Get the eeprom part number, look into the datasheet for the amount of memory it has in bytes, substract 1 to that number and you will get the E2END value

boelle, 2018-09-10:
512 kb must be 524288,  minus 1 makes it 524287

in hex that is 7FFFF, did i get it correct?

ejtagle, 2018-09-10:
Yes 👍

thinkyhead, 2018-09-10:
UBL allows up to a 15 x 15 grid, which requires 900 bytes of storage per mesh. So with a 512K EEPROM you'll be able to store around 581 meshes.

boelle, 2018-09-10:
hehe, i was thinking of doing an "over the top" experiment with like 20x20 grid or a bit more

i know its "insane" but why not when there is space enough, but of course eeprom is one thing, the DUE needs to be able to load it too

thinkyhead, 2018-09-10:
Since the pins are almost entirely identical to RAMPS_FD_V1, you should follow the example of `pins_RAMPS_FD_V2.h` by including `pins_RAMPS_FD_V1.h` and applying only the necessary modifiers.

boelle, 2018-09-10:
yes but i wanted to get rid of a few lines and did not  know how

#define TEMP_3_PIN         -1   // fewer compiler warnings
#define TEMP_4_PIN         -1   // fewer compiler warnings

// SPI for Max6675 or Max31855 Thermocouple
#if DISABLED(SDSUPPORT)
  #define MAX6675_SS       53
#else
  #define MAX6675_SS       49
#endif

#if HAS_DRIVER(TMC2208)
  /**
   * TMC2208 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial1
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1
#endif

//
// M3/M4/M5 - Spindle/Laser Control
//
#if ENABLED(SPINDLE_LASER_ENABLE) && !PIN_EXISTS(SPINDLE_LASER_ENABLE)
  #if HOTENDS < 3
    #define SPINDLE_LASER_ENABLE_PIN  45   // Use E2 ENA
    #define SPINDLE_LASER_PWM_PIN     12   // MUST BE HARDWARE PWM
    #define SPINDLE_DIR_PIN           47   // Use E2 DIR
  #endif
#endif

ejtagle, 2018-09-11:
You can use #undef to undefine a given symbol, and then redefine its value if you need to

#undef HOTENDS
#define HOTENDS 2

boelle, 2018-09-11:
yes but the things i have removed are things that should not have been there in the first place

and using undef would just make it look stupid and unclean

that is why i made it independant from other files and started with a clean slate

thinkyhead, 2018-09-11:
This is also an acceptable approach:
```cpp
#define BOARDNAME_V3
#include "pins_BOARDNAME.h"
```
And then in `pins_BOARDNAME.h`…
```cpp
#ifdef BOARDNAME_V3
  #define DONUT_PIN 11
#else
  #define DONUT_PIN 12
#endif
```

cjsoong, 2018-09-12:
> 512 kb must be 524288 bytes (512x1024), minus 1 makes it 524287
> 
> in hex that is 7FFFF, did i get it correct?

The general i2c bus EEPROM is marked with size/Bits.
So 512k =64k bytes  @=0xFFFF

boelle, 2018-09-12:
So does marlin have 64kb of space with that setting?

thinkyhead, 2018-09-12:
> again i made an independent file for those reasons

If there are known errors, then it's best to fix those errors first, then you will be free to follow the preferred methodology.

thinkyhead, 2018-09-12:
> So does marlin have 64kb of space with that setting?

- For 64K E2ND is 0xFFFF (2^16 - 1)
- For 512K E2END is 0x7FFFF (2^19 - 1)

cjsoong, 2018-09-12:
@boelle 
Can refer to AT24C512 datasheet.
Ramps FD 2.1 is now 64K bytes EEPROM. (E2END=0xFFFF)
[image]