MarlinFirmware/Marlin #11791: Spelling Error "Leveling" vs "Levelling"

CriticalPoint, 2018-09-10:
### Description

Throughout Marlin v1.1.9, the spelling of "Levelling" is incorrectly spelt as "Leveling"

### Steps to Reproduce

#define MESH_BED_LEVELING
#define LCD_BED_LEVELING
1. Use Mesh_Bed_Leveling via the LCD

**Expected behavior:**
It works, but the OCD in me makes me audibly gag

**Actual behavior:**
You see the message "Leveling Done!"

#### Additional Information ####

Doing a find and replace can mitigate the spelling error, however it will throw an error as Arduino cannot then reference the file "mesh_bed_leveling.h", as this is also spelt wrong.

CriticalPoint, 2018-09-10:
[image]
Urgh, don't worry, I've just reali**s**ed that it's cultural

CriticalPoint, 2018-09-10:
My Friend David has something to say about all this - https://www.youtube.com/watch?v=om7O0MFkmpw

rmoravcik, 2018-09-10:
Well, you can create a new language file (language_en_uk.h) and translate strings to british english :-)

Roxy-3D, 2018-09-11:
I'm a horrible speller....   And I know I have spelling errors in my code and comments....

Oh well....

CriticalPoint, 2018-09-11:
Isn't that part of your work flow, to revisit the code once it's complete
and tidy it up, spell check et-cetera? Personally I always go through my
code once it works, I wouldn't put my name to it until its finished.
Possibly a bit different for me, being a professional, but I still believe
one should put 110% into anything they make, physical or not!

thinkyhead, 2018-09-11:
> Isn't that part of your work flow, to revisit the code once it's complete and tidy it up, spell check et-cetera?

I take it upon myself to clean up any submission that I can review before it gets merged. Part of it is having my name associated with the commits, but I also think it's important to provide consistent and readable code with concise and clear commentary for people new to the codebase who may be wanting to modify or study it academically.

That said, the HAL is currently in a process of rapid expansion and elaboration, so I'm holding off on a big cleanup and overhaul until a bit later on.

closed 2018-09-11 by thinkyhead

thinkyhead, 2018-09-12:
Well, of course his handle _is_ "CriticalPoint." 🤣

CriticalPoint, 2018-09-12:
Duped by my phones auto-correct, thanks for the edits thinky, just the right amount of cunting, I can tell we'd get along well.

Either way, the response was specifically directed at Roxy's "Oh well...", and not even maliciously, it's just that I personally can't stand the laissez-faire "don't really care that it's not finished as it works" approach. That 10% Roxy is the inferred additional effort that mitigates my previous paragraph, one might even call it pride.

Noted that I originally believed it to be a typo, because I'm stoopid and didn't realise it was spelt differently between the two adaptations of said common language. I then remembered a funny video I watched and thought "Hey, I'll post that and that might get a laugh" and then didn't think too much more about it.

I thought I'd do as rmoravcik suggested and create the language file, the easiest way being to clone the preexisting language_en.h and make the changes, but I found that in that file, the word was already spelt with a pair of L's.

[image]

Anyway, I didn't mean to trigger anyone, I appreciate the project and the efforts that people donate to the common cause, thinky you've helped me lots over the past year or so, I went through a spell (ha haaa) of searching for fairly niche issues and your name came up more than most, so for that alone I'm thankful.

And yes, you've nailed it, that is exactly why my handle contains the word, because I am extremely critical. It's nothing short of a curse.

CriticalPoint, 2018-09-12:
> thinky you've helped me lots over the past year or so, I went through a spell (ha haaa) of searching for fairly niche issues and your name came up more than most, so for that alone I'm thankful.

To be more precise I'm referring to Grbl on the whole, as opposed to Marlin specific issues

thinkyhead, 2018-09-12:
No worries, CP, and happy to help out in those niches where I can! I do like the idea of a UK English language file. If you have one started, please post and I'll see that it's added to the codebase. Because of the way Marlin works, it only needs to contain the strings that differ from US English.