MarlinFirmware/Marlin #11793: Homing to Y max - Y axis moves double distance

stevegoddard, 2018-09-10:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2367642/configuration.zip)

### Description

Hi,
I have a P3 Steel (Prusa i3 clone) that i have just rebuilt with a larger bed (200 x 300).
All was working fine. But to maximise the area of the bed I could use I decided to move the Y endstop from the min to the max position.

This has resulted in very odd behaviour:

G28 Y homes the bed correctly to the Y max endstop (bed forward). 
M114 then reports the Y position at 300 (this is correct).
G1 Y150 moves the bed 300mm backwards (effectively to the Y min position)!! It should move the bed to the halfway position.
Moving Y to any values less than 150 result in the bed moving too far and the stepper jamming.
Moving Y to any values between 300 and 150 result in the bed moving twice the requested distance - for instance: G1 Y290 moves the bed backwards 20mm rather than 10mm.

My config:

    // Specify here all the endstop connectors that are connected to any endstop or probe.
    // Almost all printers will be using one per axis. Probes will use one or more of the
    // extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
    #define USE_XMIN_PLUG
    //#define USE_YMIN_PLUG
    #define USE_ZMIN_PLUG
    //#define USE_XMAX_PLUG
    #define USE_YMAX_PLUG
    //#define USE_ZMAX_PLUG

    #define X_HOME_DIR -1
    #define Y_HOME_DIR 1
    #define Z_HOME_DIR -1

    // The size of the print bed
    #define X_BED_SIZE 200
    #define Y_BED_SIZE 300

    // Travel limits (mm) after homing, corresponding to endstop positions.
    #define X_MIN_POS 0
    #define Y_MIN_POS 0
    #define Z_MIN_POS 0
    #define X_MAX_POS X_BED_SIZE
    #define Y_MAX_POS Y_BED_SIZE
    #define Z_MAX_POS 200

Config files attached.

Thanks,
Steve

brandstaetter, 2018-09-10:
> Moving Y to any values lower than 150 (i.e. 100 or 0) result in the bed moving too far and the stepper jamming.

`#define Y_MIN_POS 0`

something does not match up here - why did you set the bed size to 300 yet only want to use 150 to 300?

stevegoddard, 2018-09-10:
@brandstaetter - I want to use the entire bed!

I have the bed defined as: 

    #define Y_BED_SIZE 300

And the usable areas:

    #define Y_MIN_POS 0
    #define Y_MAX_POS Y_BED_SIZE //i.e. 300

When i home the bed, it moves to position 300 - correct. 
I should then be able to move it to position 0 - but any movements are doubled. So if i try and move it any distance greater than 150 it hits the end of the printer and jams.

brandstaetter, 2018-09-10:
Ah I see. So please check your DEFAULT_AXIS_STEPS_PER_UNIT.

thinkyhead, 2018-09-12:
Does `M502`, `M500` (and reboot) help at all?

stevegoddard, 2018-09-12:
Thanks for the suggestion @thinkyhead - I've not tried this - but i have EEPROM disabled. I'll give it a try though.
I've been going lots of Goggling and I'm pretty sure the issue relates to my TMC2130 drivers which I have just added to X,Y and Z. But they were working fine prior to swapping the Y endstop from min to max.
I'm going to upgrade from 1.1.8 to 1.1.9 - hope to finish this tonight and see if that resolves the issue. I'll report back.

thinkyhead, 2018-09-12:
Good luck. Yes, TMC2130 stepper drivers are known to do odd things like this. But let's try to rule them out first and then if needed we can see if something else is going on.

stevegoddard, 2018-09-12:
Upgrade to 1.1.9 solved this issue.

Very odd.

Thanks for all the help!

closed 2018-09-12 by thinkyhead

gmg23, 2019-05-14:
I am having this same problem running 1.1.9 and 1.1.X bug fix. Running a new Tevo Tornado stock everything. All axis work perfectly during homing, I can move the bed from the LCD, and they operate perfectly during bed leveling. Been pulling my hair out all day. Basically an G0 or G1 command cause the Y axis to go to max position (300). After that Y is completely unrecoverable unless homed. This causes prints to literally just print in a single line on the back of the bed. Y ends up at ~280 after bed leveling and as soon as the print tries to send the bed back to center, it goes to 300 and locks up. Terminal shows the print working normally. The LCD reports the Y position as if the bed is moving appropriately. I am at a total loss. Would really appreciate any suggestions. Thanks in advance.