MarlinFirmware/Marlin #11794: Marlin 2.0 MKS Sbase Diamond hotend mixing extruder

budiman7x, 2018-09-10:
hi, i have a problem when activated mixing extruder, the all three extruder behave crazy, like a 10 times moving so fast. if mixing deactivated using single extruder, it is perfect so fine

i am using titan extruder clone and drv8825 so my step is 800, just perfect, but if activated mixing extruder, i cant calibrate my step mm extruder, even changing number doesnt affect. first layer can get 2.35 mm thick, OMG

i am attach file my config for mixing extruder

[Configuration diamond.zip](https://github.com/MarlinFirmware/Marlin/files/2367849/Configuration.diamond.zip)

if using single extruder all is fine

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2367855/Configuration.zip)

[image]
[image]
[image]

coolhoun, 2018-09-10:
I am noticing perhaps the very same issue, though my setup is different.  On 1.1.9 only (not 1.1.8 or any prior built back to 1.1.0), my Diamond hotend (mixing extruder) extrudes, perhaps, three times as much as it should?  I haven't measured, but it is quite a bit more.   I went through configurations in 1.1.9 really closely, steps per mm are the same as I've always used (97), I don't need or have enabled distinct E factors.  I tried disabling a couple of the new 1.1.9 features, but I couldn't pin the issue down...

thinkyhead, 2018-09-17:
Duplicate of #11310

closed 2018-09-17 by thinkyhead