MarlinFirmware/Marlin #11795: Autostart if no screen

dagorel, 2018-09-10:
Hi all,

I am trying to migrate from 1.1.0 RC6 to 1.1.9 and it seems there was some dev done around autostart.
Indeed, I blamed cardreader.cpp and it seems some cleaning has been done between 1.1.8 and 1.1.9.
Actually, the most important thing I've noticed is the implementation of a new method called beginautostart() and stuff around (index initialized to -1,...).

As far as I understand, it breaks backward compatibility with no screen printers.

What do you think?
- Am I wrong?
- Has it been done on purpose?
- Is it a regression?

Regards,
Orel

thinkyhead, 2018-09-12:
Looks like an oversight. Put this at the end of the `setup()` function and see whether it solves the issue:
```cpp
#if ENABLED(SDSUPPORT) && DISABLED(ULTRA_LCD)
  card.beginautostart();
#endif
```

dagorel, 2018-09-12:
Hi @thinkyhead ,

Yep that's exactly what i did. I will make some tests asap (when my printer will be up again ;)) and report my feedbck here.
If it works (and i am pretty sure it will), do you like me to create a pull request?

Regards,

thinkyhead, 2018-09-12:
I'm glad to hear it works for you! It's already patched. 🏆

dagorel, 2018-09-14:
Hi @thinkyhead ,

I just tested and it works fine.

But I want to point out another issue around this autostart stuff.
As far as I understand, if the printer has a screen, we have the ability to activate the autostart through the MENU_ADDAUTOSTART define (in [Configuration_adv.h](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L583)).
Unfortunately, it seems to me the autostart is now always enabled when the printer has a screen according to the changes made in [ultralcd.cpp](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/ultralcd.cpp#L5171).

I justed tested with a screen and it breaks the menu autostart feature: The print starts automatically, even if i didn't explicitly activate the autostart from the menu.

If I am right, I suggest to revert back from
```
if (old_sd_status == 2)
  card.beginautostart();  // Initial boot
else
LCD_MESSAGEPGM(MSG_SD_INSERTED);
```
to
`if (old_sd_status != 2) LCD_MESSAGEPGM(MSG_SD_INSERTED);`

What do you think?
Regards,
Orel

thinkyhead, 2018-09-17:
The correct behavior is to run `auto0.g` when the machine boots and this file is on the SD card. The item in the menu is for when you might want to also run it later.

closed 2018-09-17 by thinkyhead