MarlinFirmware/Marlin #11797: UBL "fade out" problem

dmitry-kutergin, 2018-09-10:
### Description

I noticed that many times when I have UBL enabled, after fade distance is passed during the print, prints starts to act weird, sometimes I can see that my layer height turns to be half of what it should be (around 0.07 - 0.1mm instead of 0.15), sometimes the opposite - layer height turns to be twice the the one it should be (0.3 vs 0.15). And I have screwed Z dimension and over or underextrusion.
Once I disable UBL completely my layer height is always as expected 0.15mm.

### Steps to Reproduce

1. Do UBL mesh map (15x15)
2. Do G29 J3, before print
3. Do G28
4. First layer is more or less perfect
5. The layers above fade height (I tried 10mm and 0.6mm) are sometimes corrupted.
6. There is no unusual lines in the G-code, that can cause such a weird Z-behavior
7. Steps per mm are correct before/during/after the print.

**Expected behavior:** No layer-height corruption

**Actual behavior:** Layers are corrupted after "fade" distance"

thinkyhead, 2018-09-12:
What version of Marlin are you referring to?

dmitry-kutergin, 2018-09-12:
It is bugfix-2.0.x from week and a half ago, I will include git hash later.

thinkyhead, 2018-09-12:
What does your mesh output look like (`M420 V`)?

dmitry-kutergin, 2018-09-19:
Sorry my printer is disassembled now, for cooling rework. I will post in a few days.

dmitry-kutergin, 2018-09-21:
Here is my mesh:
Send: M420 V
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  0,295)                                                                                                      (295,295)
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv: 14 | -0.064  -0.103  -0.142  -0.207  -0.243  -0.258  -0.292  -0.293  -0.297  -0.312  -0.306  -0.272  -0.238  -0.204    .
Recv:    |
Recv: 13 | -0.064  -0.103  -0.142  -0.207  -0.243  -0.258  -0.292  -0.293  -0.297  -0.312  -0.306  -0.272  -0.238  -0.204    .
Recv:    |
Recv: 12 | -0.019  -0.071  -0.123  -0.164  -0.190  -0.195  -0.247  -0.230  -0.247  -0.248  -0.247  -0.207  -0.175  -0.143    .
Recv:    |
Recv: 11 | +0.003  -0.052  -0.107  -0.132  -0.153  -0.183  -0.186  -0.210  -0.207  -0.205  -0.203  -0.186  -0.140  -0.094    .
Recv:    |
Recv: 10 | -0.026  -0.073  -0.120  -0.142  -0.153  -0.163  -0.173  -0.176  -0.159  -0.176  -0.173  -0.137  -0.099  -0.061    .
Recv:    |
Recv:  9 | -0.073  -0.098  -0.123  -0.147  -0.145  -0.145  -0.168  -0.145  -0.125  -0.137  -0.130  -0.101  -0.069  -0.037    .
Recv:    |
Recv:  8 | -0.066  -0.088  -0.110  -0.123  -0.133  -0.127  -0.122  -0.116  -0.099  -0.102  -0.092  -0.065  -0.030  +0.005    .
Recv:    |
Recv:  7 | -0.092  -0.100  -0.108  -0.118  -0.123  -0.099  -0.112  -0.092  -0.074  -0.070  -0.060  -0.038  -0.007  +0.024    .
Recv:    |
Recv:  6 | -0.107  -0.110  -0.113  -0.127  -0.125  -0.091  -0.098  -0.072 [-0.043] -0.048  -0.030  -0.003  +0.022  +0.047    .
Recv:    |
Recv:  5 | -0.066  -0.083  -0.100  -0.096  -0.091  -0.060  -0.053  -0.041  -0.022  -0.016  -0.003  +0.015  +0.027  +0.039    .
Recv:    |
Recv:  4 | -0.062  -0.062  -0.062  -0.073  -0.062  -0.041  -0.035  -0.013  +0.009  +0.002  +0.012  +0.022  +0.040  +0.058    .
Recv:    |
Recv:  3 | -0.088  -0.088  -0.072  -0.075  -0.051  -0.028  -0.016  +0.005  +0.035  +0.032  +0.037  +0.052  +0.053  +0.054    .
Recv:    |
Recv:  2 | -0.082  -0.082  -0.012  -0.052  -0.031  +0.009  +0.010  +0.028  +0.049  +0.043  +0.042  +0.062  +0.018  +0.018    .
Recv:    |
Recv:  1 | -0.093  -0.093  -0.036  -0.011  +0.019  +0.047  +0.050  +0.069  +0.087  +0.077  +0.061  +0.039  -0.043  -0.043    .
Recv:    |
Recv:  0 | -0.093  -0.093  -0.036  +0.030  +0.069  +0.085  +0.090  +0.110  +0.125  +0.111  +0.080  +0.039  -0.043  -0.043    .
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv:     (  0,  0)                                                                                                      (295,  0)
Recv: 
Recv: Mesh is invalid
Recv: Storage slot: 1
Recv: echo:Bed Leveling On
Recv: echo:Fade Height 0.60
Recv: ok

dmitry-kutergin, 2018-09-21:
Just in case:
Send: G29 W
Recv: Unified Bed Leveling System v1.01 active.
Recv: Mesh 1 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 0.6000
Recv: # of samples: 210
Recv: Mean Mesh Height: -0.086276
Recv: Standard Deviation: 0.099073
Recv: zprobe_zoffset: -3.7999999
Recv: MESH_MIN_X  (0)=0
Recv: MESH_MIN_Y  (0)=0
Recv: MESH_MAX_X  (295 - (0))=295
Recv: MESH_MAX_Y  (295 - (0))=295
Recv: GRID_MAX_POINTS_X  15
Recv: GRID_MAX_POINTS_Y  15
Recv: MESH_X_DIST  21.07
Recv: MESH_Y_DIST  21.07
Recv: X-Axis Mesh Points at: 0.000  21.071  42.143  63.214  84.286  105.357  126.429  147.500  168.571  189.643  210.714  231.786  252.857  273.929  295.000
Recv: Y-Axis Mesh Points at: 0.000  21.071  42.143  63.214  84.286  105.357  126.429  147.500  168.571  189.643  210.714  231.786  252.857  273.929  295.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: ok
[...]

swilkens, 2018-10-23:
I can't help but notice:

`Recv: Mesh is invalid` 

That's output of `M204 V` which itself it a result of this evaluation of the mesh:

https://github.com/MarlinFirmware/Marlin/blob/ed86550061ab94f10dd1d310c994bd74db582740/Marlin/ubl.h#L357

> inline static bool mesh_is_valid() {
>       for (uint8_t x = 0; x < GRID_MAX_POINTS_X; x++)
>         for (uint8_t y = 0; y < GRID_MAX_POINTS_Y; y++)
>           if (isnan(z_values[x][y])) return false;
>       return true;

`isnan()` determines if a part of the mesh is invalid, which apparently happens in this case?

dmitry-kutergin, 2018-10-23:
I suspected it was just a bug of the output, since rows and column numbers showed 15x15. Anyway I moved to 1.18 version due to constant layer shifting issue. On 1.18 I do not see this problem.

ManuelMcLure, 2018-10-24:
> I suspected it was just a bug of the output, since rows and column numbers showed 15x15.

Each row of the mesh your `M420 V` showed has only 14 valid values - the last one is "." which means an invalid point.

closed 2019-02-19 by thinkyhead