MarlinFirmware/Marlin #11799: MKS SBase - Hot end Temperature swings erratically durring Bed Probe

jfint, 2018-09-11:
### Description

My startup script goes:
G28		; Home All
G29		; Auto_Level
G92 E0		; zero extruder position
G1 X0 Y0 F6000
G4 S10
G1 X5 Y250 F6000	; move to extruder prime location
G1 Z 0.08  F1200	; Z for prime
G1 X5 Y270 E30 F300	; prime extruder
G1 X5 Y290 F9000	; wipe
G92 E0		; zero extruder position

I inserted the dwell to try to mitigate the temperature swings shown in the picture.  This seems to be happeneing durring the probe with the BLTouch, does the probe interrupt the frequency of hte PID and as such this is an expected behavior?

Git checkout on 09/06/2018

#### Additional Information
[image]

kAdonis, 2018-09-11:
Do you have #define PROBING_HEATERS_OFF enabled in Configuration.h?
```
/**
 * Enable one or more of the following if probing seems unreliable.
 * Heaters and/or fans can be disabled during probing to minimize electrical
 * noise. A delay can also be added to allow noise and vibration to settle.
 * These options are most useful for the BLTouch probe, but may also improve
 * readings with inductive probes and piezo sensors.
 */
//#define PROBING_HEATERS_OFF       // Turn heaters off when probing
```

jfint, 2018-09-11:
Yes I have, here is my config, I thought I had attached it, but failed
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2371660/Configuration.txt)

jfint, 2018-09-11:
I understand it is commented out in my configuration file during this

jfint, 2018-09-14:
I merged in the latest cod ethis morning, and this afternoon this actually progressed into a temperature overrun today.  I plant to actually turn off Heaters when probing because it looks like the PID loop does not run continiously, leaving the heater in a persistent on or off state untill it does run.  I think that turning off the heater and then udin ghe dwell to let if come back up is much safer so I'll be trying that next.

Looks very dangerous

boelle, 2019-02-20:
@jfint still having issues? much have happend since

jfint, 2019-02-20:
I have replaced my temp sensor since this time and have turned the heaters off during probing, and I'm solid with those changes.  My printer has been working every single day for the last month on some projects both paying and pleasure, so I have not had any time to try and update to the latest code on this branch.

boelle, 2019-02-20:
i also probe cold, just have to adjust the probe offset so first layer is perfect

but i get far better results when doing it that way

hope you find time to test latest bugfix

boelle, 2019-03-13:
@jfint should we put this one to sleep by clicking the close button below?

closed 2019-05-16 by thinkyhead