MarlinFirmware/Marlin #11800: Home during printing only x and y IF STEPPER MISS STEPS !!!!!!!!!

KARTHICKROBO, 2018-09-11:
IF STEPPER MISS THE STEPS MEANS ALL OUR TIME AND MONEY WASTED

SO I ASK AN FUNCTION TO SOLVE THIS PROBLEM

SOLUTION IS 

PLS GIVE FUNCTION TO HOME ONLY X AND  Y AXIS DURING RUN TIME , PRINTING TIME 

IF I FEEL PRINTER MISSED STEPS MEANS I CAN HOME X&Y AXIS SO THE MISTAKE   IS CORRECTS BY THE WAY 

THANK U

AnHardt, 2018-09-11:
Pssssst! Don't be that loud. Kids are sleeping.
You an always send an `G28 XY`.
But we can't detect skipped steps without position feedback.
If we could, there where better things we could do, without homing.

SJ-Innovation, 2018-09-11:
First. Dont shout. It wont make us add this any faster. 

Secondly, ive got a PR coming that adds a bunch of stuff to the LCD Tune menu while printing.... Including homing....

Third, Homing every minute during a print isnt a function the printer does automatically. Add it to the gcode with a post-processor if you want to do it that way. (Please dont its a terrible idea)

Also you may want to check out my thread on closed loop controllers if its bothering you so much. #11772

thinkyhead, 2018-09-12:
Duplicate of #8765

closed 2018-09-12 by thinkyhead