MarlinFirmware/Marlin #11801: Layer shifting and curious gcode execution

murdock62, 2018-09-11:
Hello everyone.
First of all, sorry for my bad english.

Following the update in version 1.1.9 on two of my printers (2 totally identical machines with A4988 drivers) I noticed 2 curious things:

	 - My startup sequence contains the following gcode:
		```
                M82
		M106 S0
		G28 ; home all axes
		G1 Z15 F3000 ; lift
		G1 X0 Y0 F7800 ;
		M140 S50 T5 ;
		M190 S50 T5 ;
		M104 S200 T0 ;
		M109 S200 T0 ;
		G92 E0 ; reset extrusion distance
		G1 F1500 E-5 ;
		G1 F200 E5;
		G92 E0 ; reset extrusion distance
		M117 Clean nozzle ;
		M300 S300 P100 ; 
		G4 P500 ; 
		M300 S300 P100 ; 
		G4 P500 ; 
		M300 S300 P100 ;
		G4 P5000 ;
		...
```
		
		
		The goal is to heat the bed, then the nozzle and once at a temperature to drop the pressure in 
                the nozzle and then emit 3 beeps spaced 500ms to the machine.
		
		
		With the version 1.1.8, on both machines no worries, it gives : Heats, retracts 5mm, extrusion 
                5mm, message display then beep(500ms)beep(500ms)beep, wait 5S and start printing.
		
		With the version 1.1.9, on both machines, it gives: Heats, retracts of 5mm, display of the 
                message, beep extrusion of 5mm, then beep beep, wait of 5S and departure in printing.

		If we only rely on what we hear, it makes beep, we hear the end of the extruder movement 
                with a very marked pause of almost 1 to 1.5s and the last two beep with the correct spacing 
                for them (500ms).

	- Second thing, as I said before, my two machines are totally identical with A4988 drivers.
	        With version 1.1.9, one of the machines prints normally, but the other causes the Y axis to shift 
                out of layers. (Problem seems to be known on this version)   
                I did not activate the S_CURVE function.

I hope this can help ...

Best regards,

murdock62, 2018-09-20:
Nobody to explain the problem of beeps please ??

thinkyhead, 2018-10-02:
Download and flash the latest `bugfix-1.1.x` branch, which is a bit more recent than 1.1.9, and after flashing the firmware and before testing, do `M502`, `M500` to ensure that the EEPROM is initialized.

Please ZIP and attach your configuration files to your next reply.

closed 2018-11-14 by thinkyhead

murdock62, 2019-01-13:
Hi sorry for this late reply but many works this months.

No layer shift with the latest bugfix-1.1.x !!! so very good news !

But for the first issue same thing :

With the version 1.1.8, on both machines no worries, it gives : Heats, retracts 5mm, extrusion
5mm, message display then beep(500ms)beep(500ms)beep, wait 5S and start printing.

With the version 1.1.9, on both machines, it gives: Heats, retracts of 5mm, display of the
message, beep extrusion of 5mm, then beep beep, wait of 5S and departure in printing.

If we only rely on what we hear, it makes beep, we hear the end of the extruder movement
with a very marked pause of almost 1 to 1.5s and the last two beep with the correct spacing
for them (500ms).
The printer work very well but it's very curious ...

Many thanks for your answer.

Best regards,