MarlinFirmware/Marlin #11804: RAMBO board problem with extruder fan & part cooling fan

sigourosandreas, 2018-09-11:
### Description

Marlin 1.1.9 gives an error when I define E0 auto fan. Also after uploading the program, the part cooling fan is not working. The board is RAMBO. In previous vertion (1.0.2) was working fine. If I set different pins, is working.
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2371249/Configuration_adv.txt)

### Steps to Reproduce

I have defined in marlin 1.1.9 the RAMBO board in configuration.h file
      #define MOTHERBOARD BOARD_RAMBO
I use the configuration.adv.h, the one I have attached here.
For extruder fan I use the FAN0 possition which is pin 8 and for part cooling fan I use FAN2 possition which is pin2.
I upload and gives the following error.
      SanityCheck.h:1043: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
      #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
This error is not right because I am not using E0_AUTO_FAN_PIN equal to FAN_PIN.

Second mistake is that the part cooling fan is not working at all as it should be.
For this, I undefine in the configuration.adv.h file the extruder fan, so it uploads the file.
After uploading I give the M106 command, the fan on the LCD is spinning but the actual fan is not. 
All connections are correct, because I was using marlin 1.0.2 with no problem.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

MrAlvin, 2018-09-14:
I Marlin 1.1.9 the fan pins for Rambo are defined as follows

#ifndef FAN_PIN
  #define FAN_PIN           8
#endif
#define FAN1_PIN            6
#define FAN2_PIN            2
(see file pins_RAMBO.h) 

In you configuration_adv.h file, you have the following definition
#define E0_AUTO_FAN_PIN 8

This makes  E0_AUTO_FAN_PIN  == FAN_PIN  == 8

------

I Marlin 1.0.2 the fan pin for Rambo are defined as follows
#define FAN_PIN            8

(see location "MB(RAMBO)" in file pins.h)
-------

Did you have any ..._AUTO_FAN_... pins defined in configuration_adv.h under Marlin 1.0.2? 

If you want to enable an  ..._AUTO_PIN_... function under Marlin 1.1.9, it would seem that you 
- either need to use any other pins besides 2, 6, 8
- or you need to set the Fan pins in file "pins_RAMBO.h" to -1, and then use those numbers in configuration_adv.h

Okay?

sigourosandreas, 2018-09-22:
Hi MrAlvin,

In marlin1.0.2 I had it as follows:

// Extruder cooling fans
// Configure fan pin outputs to automatically turn on/off when the associated
// extruder temperature is above/below EXTRUDER_AUTO_FAN_TEMPERATURE.
// Multiple extruders can be assigned to the same pin in which case
// the fan will turn on when any selected extruder is above the threshold.
#define EXTRUDER_0_AUTO_FAN_PIN    8
#define EXTRUDER_1_AUTO_FAN_PIN   -1
#define EXTRUDER_2_AUTO_FAN_PIN   -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

I think that something is wrong in marlin 1.1.9, because the settings are the same as in the previous version.

________________________________
Από: MrAlvin <notifications@github.com>
Στάλθηκε: Σάββατο, 15 Σεπτεμβρίου 2018 1:31 πμ
Προς: MarlinFirmware/Marlin
Κοιν.: sigourosandreas; Author
Θέμα: Re: [MarlinFirmware/Marlin] RAMBO board problem with extruder fan & part cooling fan (#11804)

I Marlin 1.1.9 the fan pins for Rambo are defined as follows

#ifndef FAN_PIN
#define FAN_PIN 8
#endif
#define FAN1_PIN 6
#define FAN2_PIN 2
(see file pins_RAMBO.h)

In you configuration_adv.h file, you have the following definition
#define E0_AUTO_FAN_PIN 8

This makes E0_AUTO_FAN_PIN == FAN_PIN == 8

________________________________

I Marlin 1.0.2 the fan pin for Rambo are defined as follows
#define FAN_PIN 8

(see location "MB(RAMBO)" in file pins.h)

Did you have any ...AUTO_FAN... pins defined in configuration_adv.h under Marlin 1.0.2?

If you want to enable an ...AUTO_PIN... function under Marlin 1.1.9, it would seem that you

  *   either need to use any other pins besides 2, 6, 8
  *   or you need to set the Fan pins in file "pins_RAMBO.h" to -1, and then use those numbers in configuration_adv.h

Okay?

MrAlvin, 2018-09-22:
Well, in 1.0.2 it is possible to both have a pin defined as an EXTRUDER_0_AUTO_FAN_PIN  in configurations_adv.h _and_  the same pin defined as FAN_PIN in file pins.h
This is no longer possible under Marlin 1.1.9. 

With Marlin 1.1.9 you must choose if a pin number can be used as a FAN_PIN in the pins_RAMBO.h filer _or_ used as a E0_AUTO_FAN_PIN in the file configuration_adv.h

I suppose it is in order to prevent that the two methods starts to compete about who gets to control the fan. 
So what may feel like a fault, because this configuration was possible before, but it is most likely a new and improved safety feature with Marlin 1.1.9

Either way, to get the E0_AUTO_FAN_PIN to work, with Marlin 1.1.9 you must (currently) set FAN_PIN to -1 in the file pins_RAMBO.h

sigourosandreas, 2018-09-23:
Ok, I understand what you mean.

The thing that in this case I will dissable one port/connector on the board, the FAN 0 connector.
It doesn't sound right. 

I have connected my extruder in ports/connectors EXTRUDER 0, which are HEAT 0 and FAN 0. By setting FAN_PIN to -1 in the file pins_RAMBO.h, FAN 0 will no longer be working. Isn't this strange?
[image]

MrAlvin, 2018-09-23:
hmmm....  E0_AUTO_FAN_PIN to 8, the fan should hopefully start up, whenever it is deemed necessary. I suppose this is the intended use and purpose of the Auto_Fan functionality. 

And because you want the pin 2 to be the part cooler fan, I think you need to set FAN_PIN to 2. 

This way the Firmware will associate the two pins, with the same hot-end/extruder. 

An easier way might be to use the motherboard FAN_0 as the part cooler, and the FAN_2 as the extruder auto cooler.  This way you might not need to change so many pin assignments.

boelle, 2019-02-20:
@sigourosandreas problem solved?

sigourosandreas, 2019-02-20:
No, they have changed something in marlin since v1.0.2 and it is not working properly with rambo board.

boelle, 2019-02-20:
when did you test last ?

sigourosandreas, 2019-02-20:
It must be 3-4 months back.
Is there something changed?

boelle, 2019-02-20:
yep, about every 2 days bugfix 2.0 is updated

sigourosandreas, 2019-02-20:
Aaaa, ok.
I will have a look.
Thanks

InsanityAutomation, 2019-02-20:
I have this feature working on a couple machines. The auto fan pin should not be assigned to anything else, so comment the line out. Part cooling fan should be the first fan defined.

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

acestu, 2019-03-30:
Hi, I have just built an annet A8 and it was printing fine, after flashing Marlin 1.1.9. with the example A8 config files when I try to print the Part Cooling Fan does not auto operate during printing, could this be a wrong pin issue ?
thanks

Bob-the-Kuhn, 2019-03-30:
The E0_AUTOFAN option is disabled by default.  In configuration_adv.h set E0_AUTO_FAN_PIN to the fan pin.

You may need to set FAN_PIN to an unused pin so that the two are not fighting for control.

acestu, 2019-03-31:
Thanks Bob,

In 1.1.9. I have Power-Loss-Recovery enabled,  After stopping a print, it stops but then the Power-Loss-Recovery menu comes up on the LCD (possible bug)

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-11 by shitcreek

ketharjp, 2020-01-13:
Hello,

I have an Airwolf3DHD for which I intend to modify the extruder head assembly and install an additional fan for part cooling as I plan to print PLA on the printer.

I'm unsure as to how to go about doing this. The printer has a RAMBo 32-bit board that runs on Marlin firmware. I have been able to obtain the original version of the configuration.h file for the printer from the manufacturer. Ideally I prefer the print cooling fan to be controlled by the 3D printing software.

Any help in this direction will be greatly appreciated. Thank you!