MarlinFirmware/Marlin #11805: [2.0.x] USE_CONTROLLER_FAN no error when trying to use PWM on non-pwm pin (Arduino Mega ramps 1.4)

hamster65, 2018-09-11:
I have set E0_AUTO_FAN_PIN to 65 which is working fine. EXTRUDER_AUTO_FAN_SPEED is set to 255. Any other value results in an error saying that Pin 65 is not an PWM pin, which is right.

I enabled USE_CONTROLLER_FAN and set CONTROLLER_FAN_PIN to 66. If I set CONTROLLERFAN_SPEED to 100, the fan won't run at all. If I set it to 150, it runs at full speed. So PWM is not working. From what I read, Pin 66 is not a PWM pin, so there should be an error during compile time. There might be a check missing for that.

Am I correct that only PINs 2,4,5,6,7,11,12,13,44,45 and 46 can be used for PWM?

AnHardt, 2018-09-11:
On the Arduino Mega pins **2-13** and **44-46** can do **hardware** PWM.
About **every** pin can be driven with a **software** PWM. 
Try `FAN_SOFT_PWM`!

hamster65, 2018-09-12:
Thanks! When speaking of PWM, I meant "hardware PWM". I used pins 4 and 44 for both fans and it works fine now with hardware pwm. I do not want to use software PWM if it can be avoided.

Still I think that the setup described above (using pin 66 for CONTROLLER_FAN_PIN and setting CONTROLLERFAN_SPEED != 255) should give an error when compiling because hw PWM is not possible on PIN 66.

boelle, 2019-02-20:
@hamster65 problem solved?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead