MarlinFirmware/Marlin #11809: Question/Suggestion: Should the Change SD card menu item use M21 or M22

gloomyandy, 2018-09-12:
Currently the manual SD change menu item (present when no sd card detect pin is defined) simply issues an M21 command which basically re-initialises the SD card. But would it be better if instead it used the M22 command to release the sd card and return the system to the initial no sd card present state, with the "Init SD" card menu item present and ready to initialise/read a new card? At the moment it seems that the SD Change menu item is really just a duplicate of the refresh command present in the print from sd menu. As a result there is no way to tell Marlin to stop accessing the card so that it can be safely removed. Is there a reason for the current use of M21 rather than M22?

One reason why I'm asking about this is that it is now possible to share an SD card between Marlin and a PC via USB. It would be nice to have a clear cut way to control which system had "control" of the SD card to ensure that both systems do not attempt to modify the file structure at the same time.

forkoz, 2018-09-14:
Wouldn't that mean then that change SD as it is doesn't really change the sd?

gloomyandy, 2018-09-14:
It sort of does in that if you have removed the old SD card (without Marlin knowing since this menu is only present if you do not have SD detect pins), and inserted a different SD card then change SD will tell Marlin to read the new SD. So at the moment it is a way to say to Marlin "the SD card has changed". But in some situations I think it would be better to have it tell Marlin "I am about to change the SD card" and then use "Init SD" to tell it when the new card has been inserted, which is what using M22 would do.

comps, 2018-09-14:
Looking at the code, `sd_status` seems like it could change during runtime, so you can probably connect an arbitrary free pin via a patch cable to `SD_DETECT_PIN` and use M42 to "eject" the card.

I don't have a 32bit board yet, but I've read about Marlin exporting the SD card as mass storage via the (native) USB of an Arduino DUE (presumably at very low speeds). I may remember wrong, though.

forkoz, 2018-09-14:
If you have an SD_detect pin I assume you would never see this or have an issue. We can't really use the pin work around for those onboard SDs.

Bergerac56, 2018-10-26:
I fully agree with @gloomyandy .

For those who are using mainly (only?) the onboard card to change firmwares AND print gcode files, we need a way to tell marlin to point to the onboard sd card when printing a file or to restate the USB communication to the sd card (USB card reader) to write new files or firmwares.

So, in other words, when we want to print using the onboard sd card, we "init" sd card. And when we want to write a new file to the onboard card or write a new firmware, we "change" sd card. (Btw. The label is a bit confusing for me, not english native speaker.  I changed it by "Release SD Card")

This implies, as Andy said above that M21 is changed by M22.

But this is only my 2 cents...

thinkyhead, 2018-11-04:
"Release" makes the most sense to me. Simulating the removal of the SD card from the slot.

gloomyandy, 2018-11-04:
@thinkyhead any thoughts on how to move forward on this? Perhaps we should have two options one uses "Change SD card" and M21 and is used to tell Marlin that you have inserted a different SD card (essentially what we have now and what people are used to). The other option has "Release SD card" and M22 and is used for the situation I've described above where we want to switch between Marlin access and PC access to the same card, or can be used by those folks that expect to "eject" the card before you remove it (which arguably everyone should be doing). Or would you be happy just to change it to "Release SD card" and M22? After all the "Print from SD" menu contains a refresh option which is essentially just the same as using the current "Change SD card".

thinkyhead, 2018-11-05:
I think a single item which says "release" when the SD card is properly initialized, and which says "change" when the SD card is not-detected (and/or released) would be adequate. This will simply mean that if you remove the SD card without "releasing" first, you'll have to click the item twice to get the card to be recognized.

All of this will of course only apply when there's no `SD_DETECT_PIN`.

boelle, 2019-02-20:
@thinkyhead should this not be converted to a feature request?

gloomyandy, 2019-02-20:
I think the solution mention above by @thinkyhead sounds good. I would say it is still an issue as there is currently no safe way to remove an SD card using the menu.

boelle, 2019-02-20:
is it not a problem on systems that write to the card?

since marlin only reads the issue should not be there, but i could be wrong

InsanityAutomation, 2019-02-20:
Marlin writes with power loss recovery on when printing from SD and with file transfer from remote host to SD.

boelle, 2019-02-20:
of course, but i would not use that, will wear out the card to quickly, but that is just me

thinkyhead, 2019-02-20:
The plan is to replace "Change SD" with "Release SD" / "Insert SD" based on whether Marlin knows there's an SD card inserted. When there's no SD detect pin, Marlin will just use a flag.

closed 2019-02-20 by thinkyhead