MarlinFirmware/Marlin #11810: [2.0.x, Question] Does LINEAR_ADVANCE support dual extruders?

marcio-ao, 2018-09-12:
Does LINEAR_ADVANCE currently support dual extruders? We are seeing some fairly bad dual-prints when that feature is enabled, although it looks fine on a single extruder hotend. Just curious whether LINEAR_ADVANCE takes into account the possibility that you might have multiple extruders.

Bluelinegecko, 2018-09-12:
It looks like someone was already working on this and merged it in a few days ago in this thread #11789

marcio-ao, 2018-09-12:
I guess that thread is about using different K factors per nozzle. So I guess it is supposed to work fine with duals? In my particular case, both nozzles have the same K factor. It just wasn't sure to me whether the algorithm in Marlin had been implemented to take into account two nozzles (in which case each nozzle would need its own set of state variables, as the pressure in each would be different)

marcio-ao, 2018-09-12:
Attached is a photo of what we are seeing on a dual print -- on the right side there are alternating layers of over and under extrusion. The left print was done with LIN_ADVANCE and JUNCTION_DEVIATION off, while the right print was done with LIN_ADVANCE and JUNCTION_DEVIATION on. Though I have not tested both features independently, I suspect LIN_ADVANCE as the cause of the problem.

The material is PLA on both nozzles, so independent K factors should not be needed.

[image]

margaale, 2018-09-12:
Yesterday i saw the something similar in my printings. Just finished installing a chimera an i saw an overextrusion on the second nozzle. Both are well calibrated on their own. Printing with the second nozzle only works as usual...

Sebastianv650, 2018-09-13:
As long as we are talking about classic dual nozzle systems and not mixing ones it should work. But I was never able to test it as I don't have a multi nozzle printer.
Some questions:
- A print using only nozzle 1 is OK?
- A print using only nozzle 2 is OK?
- When both nozzles are used in a print but one nozzle is only used up to some height below the max. print height, does the print looks OK above this level where only one nozzle is working again?

marcio-ao, 2018-09-13:
Just to update this post, we were able to get a good print with just LINEAR_ADVANCE turned on. So perhaps it is a combination of LIN_ADVANCE and JUNCTION_DEVIATION, or it actually is JUNCTION_DEVIATION that is causing a problem.

margaale, 2018-09-13:
> A print using only nozzle 1 is OK?
A print using only nozzle 2 is OK?
When both nozzles are used in a print but one nozzle is only used up to some height below the max. print height, does the print looks OK above this level where only one nozzle is working again?

- Yes
- Yes
- No

> Just to update this post, we were able to get a good print with just LINEAR_ADVANCE turned on. So perhaps it is a combination of LIN_ADVANCE and JUNCTION_DEVIATION, or it actually is JUNCTION_DEVIATION that is causing a problem

Mine has the same settings. Gonna try this weekend all the posible combinations

closed 2019-02-14 by marcio-ao