MarlinFirmware/Marlin #11813: Marlin 1.1.9 - Bed temp goes to 0 when going in Tune from the control box

VmtSquad, 2018-09-12:
### Description

Since upgrading to 1.1.9 I'm noticing a easy to replicate behavior that occurs on 2 occasions. 

Issue: Bed temp goes to 0 

### Steps to Reproduce

1. During a print, I simply go to Tune to adjust some settings manually and I notice that the bed temp config went automatically to zero. I then go change it fast so it wont affect my print. It never happens again if I go back to Tune after that.

2. I also noticed that the same issue occurs when I use "Post Processing Scripts" with Cura. I sometimes change the fan speed or bed temp based on layer height (ChangeAtZ 5.1.1). The post processing scripts are processed like they should but the bed temp goes to 0 on the first "ChangeAtZ" script. This doesn't happened all the time and I still have to test more to replicate it when I want. 

**Expected behavior:** : the bed shouldn't change temp if not requested 

**Actual behavior:** the bed goes to 0

VmtSquad

thinkyhead, 2018-10-03:
Give the current `bugfix-1.1.x` branch a try and see if it resolves your issue.

boelle, 2019-02-20:
@VmtSquad tried what thinky suggested?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead