MarlinFirmware/Marlin #11815: Can G-code scripts be run automatically on inserting an SD card when using Marlin Firmware?

greenonline, 2018-09-12:
### Description

G-code scripts to be run automatically on inserting an SD card

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

This is a **Feature Request** not a bug report.

It would be great if I could get a printer which is *already powered on* to auto-start printing when the SD card is inserted, that is to say, that the SD card insertion event triggers the printing to commence.

This is different from the existing functionality of automatically running a G-Code file *only* when *powering on* the printer with a SD card already present, which contains a auto0.g file.

Note that with this request, it would be better if the `autoX.g` file was not (necessarily) required, but if that is not possible then the addition of a feature whereby a printer which is already powered on, and an insertion of a SD card with an `autoX.g` file present in the root which triggers printing will be fine.

**Expected behavior:** [What you expect to happen]

Upon inserting an SD card, into a printer that is *already* switched on, the printer begins to execute the G-code contained therein

**Actual behavior:** [What actually happens]

Printing does not start automatically, upon insertion of an SD card

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

I have asked about this issue on the Marlin RepRap forum (see [Can G-code scripts be run automatically on inserting an SD card when using Marlin Firmware?](https://reprap.org/forum/read.php?415,833394) for more details), and I was told to post a Feature Request here.

Apparently, to quote the reply that I got on the Marlin RepRap forum:

> I do not think that it would require much work to implement a "card.autoprintfile()" function in Marlin.
It seems to me that a call to such a function could happen at line 5172 right after:
>```
>    else
>    LCD_MESSAGEPGM(MSG_SD_INSERTED);
>```
>in the file ultralcd.ccp

thinkyhead, 2018-09-12:
I'm not in favor of this as a general feature. We already have a feature request to be able to run custom G-code in response to various events, with SD card insertion being one of those events. Would that feature serve your needs?

greenonline, 2018-09-12:
Ah ok, if this is a duplicate then fine, and apologies - I did have a look to see if this feature had already been requested prior to the Marlin forum post, but I couldn't see one. Do you have an issue number, or link, so that I can take a look and check?

Yahyaoui94, 2018-11-23:
How can I add a function (only for the printer I am working on for the moment) that allows me to print all the files on the sd card one after another? I am most interested in the automatic next file execution feature. I noticed the AUTOSTART but if I understood correctly, does it mean that if I rename my 3 files as AUTO0.g AUTO1.g AUTO2.g , they will be processed like a queue after the completion of the previous one?

closed 2018-11-30 by thinkyhead