MarlinFirmware/Marlin #11820: SanityCheck.h:352:6: error: #error "BLUETOOTH is only supported with AT90USB."

sandeepdohare, 2018-09-13:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/2377845/config.zip)

Hi
I tried uploading new Marlin firmware 1.1.9 , for which i have updated the configuration.h file , with latest arduino ide when i tried compiling the firmware i faced a compilation error .Since i don't know the programming language .I couldn't figure out what the issue ,This might be very stupid issue , Kindly guide me to resolve the issue.

Expected behavior : compilation should not have much problem

It showed the error as mentioned below 
 [In file included fromC:\Users\OPTIMUS\AppData\Local\Temp\arduino_build_740002\sketch\MarlinConfig.h:42:0,

                 from C:\Users\OPTIMUS\AppData\Local\Temp\arduino_build_740002\sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:352:6: error: #error "BLUETOOTH is only supported with AT90USB."

     #error "BLUETOOTH is only supported with AT90USB."

      ^

exit status 1
#error "BLUETOOTH is only supported with AT90USB."]

MrAlvin, 2018-09-14:
In your configuration.h, check to see if  it has: 
#define BLUETOOTH

If it does, change it to: 
//#define BLUETOOTH

closed 2018-09-17 by thinkyhead

sandeepdohare, 2018-09-17:
Thanks, Mr Alvin for your help... Finally upload the firmware and it's working...