MarlinFirmware/Marlin #11821: Using Features Available In Marlin Bugfix 2.0x

DSDiver, 2018-09-13:
At the outset a big thanks and tons of appreciation to the entire team involved in creation of this amazing software. I've been using it for a long  long time now. 

            Over a period of time a lot of new features have been incorporated in Marlin. However, using these features is another issue. Someone who is not a coder (like me) is faced with these challenges when using Marlin Bugfix 2.0x: -

1.   **List Of Features available** .... Though some of the major features are listed in some of the documentations, there's no comprehensive, updated list of all the features available. Sifting through all the Commits / Pull Requests / Issues raised and resolved it may be possible to make a list but that's a Herculean task.

2.   **How To Use The Features Available.** Assuming I make a comprehensive list of features, the next challenge faced is **HOW TO USE THESE FEATURES.** For instance, recently some new feature have been incorporated  ... for example .....
              (a)  **Resuming Print after Power Outage** .... I want to use this feature but have not been able to 
              determine the hardware changes required.
              (b)   **Compiling Marlin for MKS SBase.** Have tried various methods including using ATOM /    
              PlatformIO but failed.

I would be Grateful (and so would be so many others) if there could some guide on this issue.

AnHardt, 2018-09-16:
1.) The definitive list of all available features is `Configuration.h` and `Configuration_adv.h`
2.) If not self explaining, you can also find a very brief description there. If that is not enough try [http://marlinfw.org/](http://marlinfw.org/), google, or search in this issue tracker or the PRs. Ultimatively look up the source. Every open source project is perfectly documented by the source. If you don't understand a features theory (not implementation) - you probably don't need it.

thinkyhead, 2018-09-17:
I concur. Until we've gotten more help with bringing the documentation up to date at http://marlinfw.org the best source for information on each feature is the configuration files. Your questions about features are answered there, and the instructions for building Marlin 2.0 on boards like the SBASE are currently posted on the website.