MarlinFirmware/Marlin #11822: [2.0.x] Inconsistent extrusion

ktand, 2018-09-13:
I have an issue which seems to be caused by inconsistent extrusion. I'm running a STM32F4 based board in a Prusa clone and a RAMPS board in another Prusa clone. Both printers are using the same stepper drivers and the same velocity and acceleration parameters. `LIN_ADVANCE `is disabled but both printers are using `ADAPTIVE_STEP_SMOOTHING `and `S_CURVE_ACCELERATION`. The STM32F4 printer is running the latest bugfix-2.0.x code while the RAMPS printer is running 1.9. Prints from the RAMPS printer looks MUCH nicer.

It seems like there's some inconsistent extrusion on the STM32F4 machine. Every printed segment is smaller in the middle and wider in the ends. The width of the object is not consistent, it varies between 1.8 and 1.94mm. The "corners" are much more jaggy on the STM32F4 machine. My guess is that it overextrudes at the beginning and end of each segment.

I have attached the G-Code that was used on both machines.

Any ideas what could be causing this?

I did reduce the flow rate to make the problem more visible in the pictures.

[image]

### Pictures of prints using the STM32F4 machine

[image]
[image]
[image]

[image]

[Calibration Curve Problem.zip](https://github.com/MarlinFirmware/Marlin/files/2380480/Calibration.Curve.Problem.zip)

ModMike, 2018-09-13:
I too am struggling with inconsistent extrusion on my Azteeg X5 Mini Wifi.

Similar issues as above and crazy ABS multiplier is at .84 just to get right wall thickness.  I also have LIN_ADVANCE disabled and S_CURVE_ACCELERATION enabled.

I tried with and without S-Curve, same issue.

I've redone the config.h twice from scratch.

boelle, 2019-02-21:
@ktand is this still a problem when using latest bugfix 2.0?

ktand, 2019-02-23:
This is definately better now. I'll close this issue.

closed 2019-02-23 by ktand