MarlinFirmware/Marlin #11823: Marlin 2.0 sensorless homing on Delta

grieml, 2018-09-13:
Hello everyone, 

yesterday I tried Marlin 2.0 on my delta machine. I use the einsy rambo and also enabled sensorless homing:

I played around with the sensitivity but the Z axis keeps crashing. X and Y axis homing works fine. Any Ideas what to do?

Same issue with Marlin 1.1.9

Thanks for your help!

tonokip, 2018-09-20:
In pins_EINSY_RAMBO.h on line 70 replace:
#define Z_STOP_PIN     10

with:
#define Z_STOP_PIN     Z_DIAG_PIN

grieml, 2018-09-26:
it worked! Thank you very much - saved me a lot of time!

thinkyhead, 2018-09-27:
@tonokip — Do you recommend that as a permanent change to the pins file?

boelle, 2019-02-21:
@tonokip what thinky said ^^^

@grieml is this still a problem when using latest bugfix 2.0?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead