MarlinFirmware/Marlin #11824: Unknown command: "G30 X110 Y110"

arteck, 2018-09-13:
last version from Git 1.1.9

what i do 
G28 ; home all
G1 X110 Y110 ; move nozzle in center
M211 S0 ; disable software endstop
; now move your nozzle manually onto the bed such that it (strongly) grips paper.
G92 Z0 ; tell the printer that this is Z = 0
G30 X110 Y110 ; move the sensor over the center and let it trigger

and here is end

```
Recv:  T:21.50 /0.00 B:22.00 /0.00 @:0 B@:0
Send: G30 X110 Y110Recv: echo:Unknown command: "G30 X110 Y110"
Recv: okRecv:  T:21.50 /0.00 B:22.00 /0.00 @:0 B@:0
```

MrAlvin, 2018-09-14:
It would seem that the G30 command is only compiled into the firmware, if a probe has been enabled 

These probe types, should include the G30 code in the firmware
FIX_MOUNTED_PROBE
Z_PROBE_ALLEN_KEY
HAS_Z_SERVO_PROBE/BLTOUCH
Z_PROBE_SLED
SOLENOID_PROBE

Did you enable one of these probe types in configuration.h?

arteck, 2018-09-17:
that's it

closed 2018-09-17 by arteck