MarlinFirmware/Marlin #11828: EMERGENCY_PARSER and Arduino Mega (AT90)

NickRimmer, 2018-09-14:
### Description

Hello everyone! I tested EMERGENCY_PARCER feature, and I found something.

In "Configuration_adv.h" you said it doesn't work on boards with AT90USB, but it works property with my Arduino Mega (AT90). 

Probably there is mistake, and it works only on AT90 instead ch340? Or need someone, who can test it with board using CH340.

### Steps to Reproduce
1. I sent a GCODE with a few slow commands to printer
2. Right after that, I sent M112 command to printer
3. M112 command executed immediately, with "printer halted" message

**Expected behavior:** It must be executed from buffer, in some order

**Actual behavior:** But command executed immediately

#### Additional Information
Slow GCODE looks like:
```
G0 Z50 F200
G0 Z100 F200
G0 Z150 F200
G0 Z200 F200
```

In Configuration_adv.h I just uncomment `//#define EMERGENCY_PARSER`

AnHardt, 2018-09-15:
> In "Configuration_adv.h" you said it doesn't work on boards with AT90USB, but it works property with my Arduino Mega (AT90).
Probably there is mistake, and it works only on AT90 instead ch340? Or need someone, who can test it with board using CH340.

Almost all Arduino boards like the Mega have USB_co_processors like the CH340, smaller AVR-chips with a native USB_port oer FTDI_chips. They convert the USB serial to a two wire TTL-serial. You can usually see the communication on the pins 0 and 1. The main processor usually communicates over this real serial interface and has no native USB_interface. A few boards do have a **main** processor with native USB-interface (AT90USB... ). The emergency command parser is not implemented for them. It's located in the driver of the *real* hardware serials receive interrupt before the content is stored in the input buffer. Because i don't have a clue about USB-drivers and because they are not part of Marlin, but of the core *I* can not do that for these processors.

> G0 Z50 F200
G0 Z100 F200
G0 Z150 F200
G0 Z200 F200

For Marlins parser this is not long lasting because if the planer buffer is not full these commands do return quite immediately. Think about testing with the 'waiting' heater commands M109, M190, probing for the levelling G29 or potentially for ever lasting M0.

boelle, 2019-02-21:
@NickRimmer is this still a problem when using latest bugfix 2.0?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead