MarlinFirmware/Marlin #11830: Homing direction incorrect (goes towards the bed) (Marlin 1.1.9)

yodor, 2018-09-15:
### Description

After bed leveling is finished (g28 g29) if you do store settings (m500) then load settings (m501) and then do auto/home the head goes down towards the bed and hit it.
I can provide video if needed.

### Steps to Reproduce

1. start bed leveling (I do 3 point) and wait for it to finish
2. call m500 then m501
3. call auto/home

**Expected behavior:** [What you expect to happen]
homing to be done correctly towards the endstops

**Actual behavior:** [What actually happens]
homing tries to go below the bed

#### Additional Information
Configuration.h/adv.h [k218.zip](https://github.com/MarlinFirmware/Marlin/files/2385692/k218.zip)

GMagician, 2018-09-15:
Maybe may be fixed by ##11827 ?

yodor, 2018-09-15:
Yes it seems to be the issue, stepper direction pins are not initialized or wrongly initialized. I'll try the fix from the corresponding commit.
Thanks.

thinkyhead, 2018-09-19:
When you've confirmed everything is good you can close this issue.

boelle, 2019-02-21:
@yodor is this still a problem when using latest bugfix 2.0?

boelle, 2019-03-06:
@Thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead