MarlinFirmware/Marlin #11834: TMC2130 configuration

Yoonutz, 2018-09-15:
Hello everyone,
I am currently building my printer and I want to have it on TMC2130 drivers. I followed this tutorial https://www.youtube.com/watch?v=AVHVCPcFHUs&t=216s letter by letter and the values I get at M122 are 0:00:00:00:00 on all drivers. I have reconfigured everything from scratch and still got these values. Please tell me what informations you need more in order to help me solve this problem. Thank you

teemuatlut, 2018-09-15:
See this first http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting

pipersw, 2018-09-27:
Did you have connected the 12V power supply ? the registers are not read until 12V is powered.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead