MarlinFirmware/Marlin #11836: Marlin 2.0 MKS Sbase v1.3 with Plataform.io

oriaj3, 2018-09-15:
### Description

### Steps to Reproduce

1. Open Plataform.io and Marlin 2.0.
2. Choose in platformio.ini "env_default = LPC1768"
3. Choose in configuration.h "  #define MOTHERBOARD BOARD_MKS_SBASE"
4. Compile. 

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

brandstaetter, 2018-09-15:
see #11833 - this should fix it

oriaj3, 2018-09-16:
I solved with these:

>   // Allow reset_stepper_drivers to access the internal set_directions method.
 

>    friend void reset_stepper_drivers();

closed 2018-09-16 by oriaj3