MarlinFirmware/Marlin #11839: Marlin 2.0 Zonestar_LCD buttons don't work on mks sbase v1.3

oriaj3, 2018-09-16:
### Description

### Steps to Reproduce

1. Conect the LCD with the board.
2. Add in pins_mks_sbase.h

> #if ENABLED(ZONESTAR_LCD)
> // #define LCD_SDSS           P0_18
>   #define LCD_PINS_RS      P0_18 //28
>   #define LCD_PINS_ENABLE  P1_31 //29
>   #define LCD_PINS_D4      P3_25 //10
>   #define LCD_PINS_D5      P0_16 //11
>   #define LCD_PINS_D6      P1_30 //16
>   #define LCD_PINS_D7      P0_07 //17
>   #define BTN_EN1          -1
>   #define BTN_EN2          -1
>   #define BTN_ENC          -1
>   #define ADC_KEYPAD_PIN   P1_23// 1

3. In configuration.h add #define ZONESTAR_LCD  

**Expected behavior:**I expected the buttons to work

**Actual behavior:** The buttons do not work.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

gloomyandy, 2018-09-16:
I think you need an ADC capable pin for the keypad, none of the pins on J8 attach to the ADC. You may be able to use another pin, however I would be very cautious about doing so as you may end up feeding a 5V signal into a pin designed for 3.3V input (which at the very least may mean it does not work, and at worst will damage the mcu). I'm not familiar with the zonestar LCD so can't really comment further on that. 

If you are considering a different display the RepRapDiscount Full Graphic Smart Controller seems to work fine with the SBase board (though for now if you want to use the SD card reader attached to the LCD you will need a custom cable, I'm working with @p3p to fix that).

oriaj3, 2018-09-16:
Thanx fíe the repito, but what pins in MKS are ADC capable pins??

gloomyandy, 2018-09-16:
Looking at the LPC1768 datasheet it seems that although the pins capable of ADC operation are 5V tolerant when used for digital I/O they may not be when used as ADC inputs. This means there is a good chance you may damage the board. I've only had a quick look at the datasheet and this is my interpretation of the footnotes to the pin configuration table. If you are thinking of just trying this then please read the datasheet for yourself and draw your own conclusions:
https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf

So unless you are very confident about what you are doing and fully understand the potential issues please do not connect that keyboard output to your control board. Not all of the LPC1768 pins will tolerate 5V and there is a chance that you have already damaged something on the board. I will say again that my reading of the datasheet says that the ADC pins are not 5V tolerant when used in ADC mode, so I would urge you not to try this unless you are really, really sure about what you are doing.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-15 by p3p