MarlinFirmware/Marlin #11852: Only one mixing extruder spins when at low speeds

roguecode, 2018-09-17:
### Description
Marlin 1.1.8.
I've enabled a mixing extruder with 2 steppers. It works correctly when extruding fast, but not slow.

Below is a test script I've been running. It heats up, sets up a mix on Tool 2, then homes.
It then draws a small square 3 times with Tool 2. The only difference between each square it does is the E values. 
Here are the totals of the extruded filament for each loop:
Square 1: 8mm - this works, both motors spin
Square 2: 6mm - this works, both motors spin
Square 3: 4mm - this doesn't work.

```
G90
M82
M104 S180 T0
M109 S180 T0

; Setup Colors
M163 S0 P0.51
M163 S1 P0.49
M164 S2

G28 ; home all axes
G92 E0

T2 ; Switch to the mixed color

; Move down and to the start pos
G1 Z0.000 F1200
G1 X9.370 Y9.370 F9000

; Do a square extruding a lot, this works
G1 E0.0000 F1080
G92 E0
G1 X-9.790 Y9.790 Z0.225 E2 F720
G1 X-9.790 Y-9.790 Z0.287 E4
G1 X9.790 Y-9.790 Z0.350 E6
G1 X9.790 Y9.790 Z0.412 E8

; Do a square extruding a bit less, this works
G92 E0
G1 X-9.790 Y9.790 Z0.225 E1.5 F720
G1 X-9.790 Y-9.790 Z0.287 E3
G1 X9.790 Y-9.790 Z0.350 E4.5
G1 X9.790 Y9.790 Z0.412 E6

; Do a square extruding even less, doesn't work
G92 E0
G1 X-9.790 Y9.790 Z0.225 E1 F720
G1 X-9.790 Y-9.790 Z0.287 E2
G1 X9.790 Y-9.790 Z0.350 E3
G1 X9.790 Y9.790 Z0.412 E4

```

So basically, under a certain speed, one of the motors doesn't spin/extrude. It isn't a specific motor either - the one that spins is the one with the higher % in the mix. So the motor that spins above is the one with 0.51 in the mix (the 0.49 motor doesn't spin).

### Steps to Reproduce
Gcode above to test.

#### Additional Information
Here is my config:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2387067/Marlin.zip)

My related Marlin settings:

`#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif`

`//#define DISTINCT_E_FACTORS`

`#define DISABLE_E false // For all extruders
#define DISABLE_INACTIVE_EXTRUDER false // Keep only the active extruder enabled.`

Thanks!

thinkyhead, 2018-09-17:
We won't be patching 1.1.8. Please test version 1.1.9 and `bugfix-1.1.x`. If neither of these versions works, we're probably not too surprised, since this is a duplicate issue.

thinkyhead, 2018-09-17:
Duplicate of #11310

closed 2018-09-17 by thinkyhead

roguecode, 2018-09-17:
Thanks for the super quick reply, and sorry for the dupe, I didn't find those.
I'll try the later 1. versions, and then 2.