MarlinFirmware/Marlin #11853: [2.x,Question] LPC1768 SW Serial support TMC2208

oakdesign, 2018-09-17:
### Description

As of PR https://github.com/MarlinFirmware/Marlin/pull/11714 with TMC2208 library 0.2.5 SW Serial on LPC1768 Palttform seems to be supported.

I'm currently Setting up a LPC1768 based Re-Arm with TMC stepper Drivers. TMC2130 in SPI either HW SPI as well as SW SPI work. Not tested in printr so far but communication.

Now by reading of the PR tried to set up TMC2208 in UART as well.
Marlin with TMC2208 Setup will compile and run on the RE-ARM and Drivers can operate (stepper move) wheras stepper comminication not working or at least RX seems not working.

From the MKS SBase Pins I took over at least the Pins Setup for a single TMC2208 Driver. 

From my reading on the LPC1768 only ins P2_10, 2_11,2_12,2_14 are Interrupt capable. So I tried with 

`#if HAS_DRIVER(TMC2208)`
`  #define X_SERIAL_TX_PIN  P3_25   // J3`
` #define X_SERIAL_RX_PIN  P2_11   // J3 Interrupt Capable`
`#endif`

And the usual Y cable with 1kOm resistor between TX-RX.
M122 repots fine for 2130 in SPI but for 2208 iget the usual
Driver registers: X = 0x00:00:00:00 indicating at least no read communication to the Driver. 

Any hints if UART communication on LC1768 specific Re-Arm will be usable?

gloomyandy, 2018-09-17:
Are you using the hardware uart? If not you may be better off using that. All pins on port 0 and port 2 are interrupt capable.

oakdesign, 2018-09-17:
No as described I'm using Software Serial. What are you referring Port 0 and Port 2 to?

gloomyandy, 2018-09-17:
What I mean is are you using the hardware uart for anything else? If not why not use it for this? As to a document...
https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf
Section 8.7.2 good enough for you?

forkoz, 2018-09-17:
Won't he run out of hardware UARTs? I thought p3p was removing software serial.. did it actually work?

gloomyandy, 2018-09-17:
Will he run out? Probably, but it depends how many TMC2208s he needs to talk to, the example above only shows one. If the hardware pins are available then using them may help to identify if there is a problem with the software serial code (the same pins should be usable for both hardware and software serial since they are on port 0). I don't really see why software serial shouldn't work but I've not really looked at the code. It seems to be present in the new framework from p3p.

oakdesign, 2018-09-18:
The example shows just one because of testing purpose. The goal is to configure full set of 5 x 2208 independent

oakdesign, 2018-09-18:
Went onwith some testing. SW Serial at least TX work. Simple to test just used 
`#define TMC_ADV() {  stepperX.shaft(true);  }`
to invert stepper dirction through 2208 GCONF register.  RX to read register still don't return values from the driver

forkoz, 2018-09-18:
Maybe there is still an issue with the library, worth asking there. The other option would be to hook something else (like TTL->serial) to the uart and see if you get anything on RX.

Afaik, there is a new one to try: https://github.com/MarlinFirmware/Marlin/pull/11792

Spawn32, 2018-09-22:
@teemuatlut 
This is the output i get from M122:

SENT: M122
READ: 		X	Y
READ: Enabled		false	false
READ: Set current	1000	1000
READ: RMS current	994	994
READ: MAX current	1402	1402
READ: Run current	17/31	17/31
READ: Hold current	11/31	11/31
READ: CS actual		0/31	0/31
READ: PWM scale		0	0
READ: vsense		0=.325	0=.325
READ: stealthChop	false	false
READ: msteps		256	256
READ: tstep		0	0
READ: pwm
READ: threshold		1601	1601
READ: [mm/s]		99.01	99.01
READ: OT prewarn	false	false
READ: OT prewarn has
READ: been triggered	false	false
READ: off time		0	0
READ: blank time	16	16
READ: hysteresis
READ: -end		-3	-3
READ: -start		1	1
READ: Stallguard thrs		
READ: DRVSTATUS	X	Y
READ: stst			
READ: olb			
READ: ola			
READ: s2gb			
READ: s2ga			
READ: otpw			
READ: ot			
READ: 157C			
READ: 150C			
READ: 143C			
READ: 120C			
READ: s2vsa			
READ: s2vsb			
READ: Driver registers:	X = 0x00:00:00:00
READ: 	Y = 0x00:00:00:00
READ: 
READ: 
READ: ok

teemuatlut, 2018-09-22:
There's nothing special needed from the library side to support LPC1768 as it only depends on an existing SoftwareSerial library and makes calls that way.
Logic analyzer would the tool of choice here but I don't know when I'll have the time to scope the communication.

Spawn32, 2018-09-22:
Ah, ok, thanks @teemuatlut :)

p3p, 2018-09-22:
I'm currently moving the LPC176x framework from Marlin into another project, so if there's an issue with the SoftwareSerial implementation I'l fix it when I get to moving it into a library,

griffin-117, 2019-01-08:
@p3p did you have any luck with this? I am experiencing the same issue with SW Serial Support on the LPC176X.

ErikFontanel, 2019-01-08:
Possibly related to #12800? I'm getting the same errors with a non-arm board if I use more than one TMC2208 driver. I also experience problems with serial comms over i2c/spi.

boelle, 2019-02-21:
if not much mistaken you cant connect more than one 2208 in uart mode
at least this should be the case with the re-arm

@thinkyhead am i correct?

gloomyandy, 2019-02-21:
This is an ongoing issue, but it is certainly possibly to connect multiple TMC2208s using hardware UARTS and there is ongoing development of a software serial device for the LPC1768 that will allow the use of multiple TMC2208s. See:
https://github.com/MarlinFirmware/Marlin/issues/13197

boelle, 2019-02-21:
so we are having multiple issues open with same "problem"?

closed 2019-03-03 by thinkyhead