MarlinFirmware/Marlin #11854: Marlin 2.0.x - LCD12864 hangs a long prints > 1day

italocjs, 2018-09-17:
I'm using the 2.0.x latest build, and on long prints the lcd hangs, no commands works, neither does the "emergency stop button" (but the printer still moving and reading gcodes).   When this happen, i noticed that the G28X at the end script doesnt work, so my nozzle will stay at the last gcode position.

Board:  Ramps 1.4
Arduino MEGA 2560
LCD: 12864 chinese clone
Marlin 2.0.x
Nobreak 1200va (so no power issues)

Happened to both printers i have, with the same firmware.

the issue happens with any gcode long enough to last more than a day, The same issue happened with v1.8, someone recommended upgrading to 2.0 to fix, the time it took to freeze was longer (before any work with more than 20h would freeze) but didnt fix.

i appreciate any help.  thx

tcm0116, 2018-09-18:
I think one seen this as well, but from just letting the printer sit idle for a long time.

italocjs, 2018-09-18:
In this picture, the lcd was not working (expected time was 49h)  
[image]

But the printing worked fine, the part finished
[image]

marcio-ao, 2018-10-15:
In our FW, we found it necessary to increase the CPU_ST7920_DELAY_2 in "ultralcd_st7920_u8glib_rrd_AVR.cpp" to make this occurrence less likely

italocjs, 2018-10-15:
Thank you a lot @marcio-ao , i'll implement this!