MarlinFirmware/Marlin #11857: Second  extruder unrecognized

Sionree, 2018-09-17:
### Description

Anet A8 with MKS Gen 1.2 card,  drv8825 drivers, Marlin 1.1.9 . Despite having defined two extruders, the second is not recognized, and moves in sync with the Z.

### Steps to Reproduce

1. I used the configuration files for A8 in the example_configuration folder

2. In Configuration.h 
   I modified:  

   #define MOTHERBOARD BOARD_RAMPS_14_EEB
   #define EXTRUDERS 2
   #define SINGLENOZZLE

3. In Configuration._adv.h 
   I checked that:

     //#define Z_DUAL_STEPPER_DRIVERS
    #if ENABLED(Z_DUAL_STEPPER_DRIVERS)
     //#define Z_DUAL_ENDSTOPS
     #if ENABLED(Z_DUAL_ENDSTOPS)
       #define Z2_USE_ENDSTOP _XMAX_
       #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
      #endif
      #endif

**Expected behavior:** 

      Driver n1 => X
      Driver n2 => X
      Driver n3 => Z
      Driver n4 => E0
      Driver n5 => E1

**Actual behavior:** 

      Driver n1 => X
      Driver n2 => X
      Driver n3 => Z
      Driver n4 => E0
      Driver n5 => Z2

#### Additional Information

[A8 v1.zip](https://github.com/MarlinFirmware/Marlin/files/2390371/A8.v1.zip)

SJ-Innovation, 2018-09-18:
Anet V1 has both of its Z steppers connected to the same driver. Just look at the board, youll see there are only 4 stepper driver chips. By connecting your extruder to the other Z port, itll be driven from the same driver as the other Z motor. Anet V1 doesnt support 2 extruders without sacrificing another axis.

Sionree, 2018-09-18:
Anet A8 with MKS Gen 1.2 card => 5 stepper :)

thinkyhead, 2018-09-19:
Your configuration files look fine.

If E1 is engaging when you move Z, and you haven't enabled `Z_DUAL_STEPPER_DRIVERS`, that implies that the E1 stepper driver is somehow physically hard-wired to the same input signals as the Z stepper driver. Otherwise it should not engage in sync with Z.

closed 2018-10-08 by thinkyhead